MarlinFirmware/Marlin #27824: [FR] ✨ A method to show the X, Y & Z position on Err: MAXTEMP: E1 screen so we can restart prints from the correct point.

a4jp-com, 2025-04-29:
### Is your feature request related to a problem? Please describe.

@thinkyhead sometimes on long 3 day prints MAXTEMP E1 errors show up but I want to be able to continue the prints instead of just throw them away by getting a readout of the final X, Y & Z points on the error screen. I'm trying to print out complicated models of proteins now.  

I asked Copilot if it had a solution and it came up with one but I just wanted to run it by you and get the feature added as soon as possible if it's okay as I'm sure this will help lots of people.
--------------------------------------------
"Copilot, how do I setup Marlin firmware so it shows the X, Y & Z position on the MAXTEMP E1 error screen?
Do something like log position while printing then when the printer crashes show last logged point."
--------------------------------------------
To display the X, Y, Z coordinates when a MAX TEMP E1 error (or other critical crash) occurs in Marlin, you need to modify the firmware to log and display the position during an error state. This isn’t enabled by default, but with the right adjustments, it can be implemented. Here's how to approach it:

Steps to Implement Position Logging on Errors

1. Enable Debug Information
- Start by ensuring Marlin's debugging and detailed feedback features are enabled in Configuration_adv.h. Look for:
#define HOST_ACTION_COMMANDS
#define HOST_PROMPT_SUPPORT

- These settings allow Marlin to communicate additional information to the host in error states.

2. Customize the Kill Function (kill())
- The kill() function is triggered during a crash, including MAX TEMP errors. To log and display the nozzle's current position (X, Y, Z) on the LCD or via the serial terminal, edit the kill.cpp file:
- Add a snippet to capture the current position and display it when an error is detected.
- Example:
void kill(const char* lcd_msg, bool steppers_off) {
  // Log current position
  SERIAL_ECHO_START();
  SERIAL_ECHOPGM("Printer Halted. Position: ");
  SERIAL_ECHO(current_position[X_AXIS]);
  SERIAL_ECHOPGM(", ");
  SERIAL_ECHO(current_position[Y_AXIS]);
  SERIAL_ECHOPGM(", ");
  SERIAL_ECHO(current_position[Z_AXIS]);
  SERIAL_EOL();

  // Display on LCD
  lcd_setstatusPGM(PSTR("ERROR: Temp E1\n"));
  char buffer[50];
  snprintf(buffer, sizeof(buffer), "X:%.2f Y:%.2f Z:%.2f",
           current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS]);
  lcd_setstatus(buffer);

  // Perform standard kill behavior
  thermalManager.disable_all_heaters();
  if (steppers_off) disable_all_steppers();
  safe_delay(1000);
}
- This ensures the position is logged to the serial console and displayed on the LCD when a critical error occurs.

3. Ensure Position Tracking Is Enabled
- The above will only work if position tracking is active. Verify that your printer's firmware is using software endstops and absolute positioning. In - The above will only work if position tracking is active. Verify that your printer's firmware is using software endstops and absolute positioning. In Configuration.h:
#define MIN_SOFTWARE_ENDSTOPS
#define MAX_SOFTWARE_ENDSTOPS

4. (Optional) Store the Position in EEPROM
- If you want the last position to persist even after a restart, you can write it to EEPROM before the shutdown:
- Add this in kill() before the shutdown:
eeprom_update_float((float*)EEPROM_OFFSET_X, current_position[X_AXIS]);
eeprom_update_float((float*)EEPROM_OFFSET_Y, current_position[Y_AXIS]);
eeprom_update_float((float*)EEPROM_OFFSET_Z, current_position[Z_AXIS]);

Key Considerations
- Thermal Safety First:
- Make sure the kill function still disables heaters immediately during temperature errors. The additional logging should not delay the shutdown process.
- Debugging and Feedback:
- Test your changes to ensure the position is displayed correctly on the LCD and sent to the serial terminal during an error.
- LCD Space Limitations:
- For small LCDs, keep the position display short and clear (e.g., X:120 Y:150 Z:0.2).

Testing
[... 112 lines cut ...]
    
    snprintf_P(buffer, sizeof(buffer), PSTR("X:%3.2f Y:%3.2f Z:%3.2f"), saved_x, saved_y, saved_z);
    ui.set_status(buffer, true);
    
    // Clear crash flag
    crash_flag = 0;
    persistentStore.write_data(EEPROM_CRASH_FLAG, &crash_flag, sizeof(uint8_t));
    
    // Optionally: Allow user to resume from this position
    // This would need additional UI interaction code
  }
}

### Are you looking for hardware support?

I'm using a Dream Maker Overlord Pro and Scoovo printer. But I'd like this on any printer I have.

### Describe the feature you want

I'd like the nozzle position saved in memory so when any error screen pops up it can be shown on the error screen. This would save lots of time. Currently we just have to measure everything with calipers or a ruler which is really bad.

### Additional context

[image]
[image]

ellensp, 2025-04-29:
This is the proverbial "Taping over a check engine light"   Ie ignore the issue and carry on...

a4jp-com, 2025-04-29:
Yes, in a way. And the error screen would be super useful by telling us where the final point was. Just now when the power is cut the only way to proceed is by turning off the printer but then we lose the location data.

I put the model data on an SD card as people have said it is safer to print that way. Especially with long prints but if we do this we are printing blind. I know from experience if we connect a USB cable and print but then accidentally open anything that connects to the printer it will restart the machine also stopping the print. In this situation it would have been great to have had the point data as well.  

Because it's a heat related error, it's probably better not to have a restart print option but at least the X, Y & Z data would allow us to buy parts and then restart from the correct point. 

As an example, my printer was set to print at 195 degrees and I saw it was mainly on 195 for 12 hours. But I did see 194 and a one time a spike of 208 around the 6 hour mark but it got past that and went back to 195. Around the 12 hour mark it might have been a tiny spike around 209 or 210 that stopped my print. I'm happy the printer is checking for dangerous temps but at the same time hate the error screen as it basically says error, all data deleted, adjust heat settings and throw away the current print T-T

a4jp-com, 2025-04-29:
I tried restarting the printer and using M114 to get a point.

>>> M114
SENDING:M114
echo:SD card ok
X:0.00 Y:0.00 Z:279.79 E:0.00 Count A:36203B:36203C:36203

Then unfortunately heated the nozzle to 150 degrees C and tried G91 then G1 Z10 F600 smashing the nozzle into a failed print. Then with another failed print G91 and G1 Z-10 F600 also smashing the nozzle into a failed print.

I though I'd just be able to move the nozzle up 400 and when one of the arms hits the endstop calculate the distance but it's more complicated than that as the start position always seems to be calculated as the top of the printer.

When the steppers are off through the menu on the printer or M18/M84 they also lose the ability to record location data :/

I can only restart the printer, home it and carefully move the nozzle through G-Code, the super slow onscreen menu or Pronterface's movement interface to get the rough location of where the print failed. 

More reasons why a stored X, Y & Z position on the error screen would be great if possible. It would save lots of time and potential damage to the printer testing stuff that might work.

thinkyhead, 2025-04-29:
Basic guidelines on Feature Requests:

1. Short Title

2. Just ask for what you want! Please don't paste code or other suggestions from AI (especially without any of the proper Markdown formatting to wrap it ?!?). The replies and code provided by these chat bots are terrible and useless and do absolutely nothing to help or guide us.

---

A "MAXTEMP" or "MINTEMP" error is a definite hardware fault — either a short or a break in the wiring. You need to fix that!!

Other types of thermal runaway might be hardware problems, but they are observation-based using a countdown timer. Anyway, we are in complete control of how to respond to these errors. They do not have to go to the `kill` screen and/or shut down the power if some user or vendor wants to modify the behavior, but that is the safest option.

It sounds like all you need is for these `kill` events to save the current position like Marlin does with standard Power Loss Recovery, and when you re-start the machine you can have the option to continue from that point.

thinkyhead, 2025-04-29:
… and please do not tag me or anyone else in Issues / Bug Reports / Feature Requests.

InsanityAutomation, 2025-04-29:
I don't believe he wants to actually directly recover / resume. I interpret this as having the positional data to find in the G-Code file to manually resume, essentially doing the operations of power loss recovery by hand. It's something I've done myself many times using octoprint logs.

a4jp-com, 2025-04-30:
Thinkthead, sorry for using the @ mark in a feature request. I just thought it was what we had to use before names in Github when posting. 

It would be great if the printer saved the current X/Y/Z position and displayed the coordinates on the screen during the kill event. A more useful kill screen. Then when we restarted the machine, an option to continue from that point, just like Marlin does with standard Power Loss Recovery, would be wonderful.

InsanityAutomation, yes, I want the X/Y/Z coordinates to be displayed on the screen after a thermal fault. If the screen is small, like on a Scoovo (16x4), it should also scroll the coordinates if needed.

InsanityAutomation, 2025-04-30:
I can almost guarantee scrolling won't happen. It'll fit static or it won't get there  in the kill state, all ISR's are stopped, screen is set 1 last time and all operations are halted.