MarlinFirmware/Marlin #27828: [BUG] Bilinear bed leveling ends only after first probe

hafunui, 2025-04-30:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I'm trying to configure marlin for a delta printer. I have bed leveling set as `#define AUTO_BED_LEVELING_BILINEAR`. I'm using `#define FIX_MOUNTED_PROBE` with `#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`. When I start the auto bed leveling process, it successfully prompts me to deploy a probe, but stops when the probe first contacts the build plate, with the probe still touching the plate, and prompting me to stow the probe. It doesn't raise the head at all, let alone any of the defined probe clearance values.

I had it working correctly on bugfix-2.1.x, where it would sample a grid, probing each point 3 times. I currently have `#define MULTIPLE_PROBING 3`, and I don't know what `EXTRA_PROBING 1` does, but I've tried it with and without, no difference.

### Bug Timeline

Had it working in bugfix-2.1.x, Cannot get it to work in 2.1.3-b2

### Expected behavior

after deploying probe, sample grid of points x number of times. Then raise head by `Z_CLEARANCE_DEPLOY_PROBE` and prompt to stow probe.

### Actual behavior

after deploying probe, makes contact with first sample point then stops. Does not raise head by `Z_CLEARANCE_DEPLOY_PROBE`. Prompts to stow probe. 

### Steps to Reproduce

Level bed
deploy probe
confirm
--stops after first probe click

### Version of Marlin Firmware

2.1.3-B2

### Printer model

Delta (Modified Anycubic Kossel Linear Plus)

### Electronics

Bigtreetech SKR-Pro 1.1

### LCD/Controller

Bigtreetech TFT35 V3.0.1

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin_2.1.3_b2_configs_hafunui.zip](https://github.com/user-attachments/files/19984950/Marlin_2.1.3_b2_configs_hafunui.zip)

ellensp, 2025-05-03:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.1.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

hafunui, 2025-05-04:
Ok, I have done as you instructed. Using `bugfix-2-1.x` (as of may 5). I didn't know about `DEBUG_LEVELING_FEATURE` before, but it provided exactly the insight I needed. I was receiving this error:

```
<<< Probe::probe_down_to_z  X0.0000 Y-96.0000 Z91.2275
 Probe fail! - Triggered early (above 19.8000mm)
```

It turns out `DELTA_HEIGHT` was set incorrectly. I had used a working value from a previous configuration, but I now have a modified print head that reduces the printable height by too large of a margin. Running `G33` delta calibration helped me gather the correct measurements, and with those, everything appears to be working correctly. 

A reminder to run G33 was even in the config files, my bad.

closed 2025-05-04 by hafunui