MarlinFirmware/Marlin #27831: [BUG] Cannot use `M84 [+ axis]` ("[axis] not disabled. Shared with [another axis]")

classicrocker883, 2025-05-03:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I thought using `M84 XYE` had always worked to disable basically any axis, but when trying to use this in a terminal, I am greeted with a message like "X not disabled. Shared with Z." or "Y not disabled. Shared with X Z."

**Configuration.adv** even says 
```y
#define SD_FINISHED_RELEASECOMMAND "M84"  // Use "M84XYE" to keep Z enabled so your bed stays in place
```

### Bug Timeline

I thought it worked at somepoint

### Expected behavior

I expect to disable any axis I wish, or all except one.

### Actual behavior

Won't let me disable a particular axis

### Steps to Reproduce

1 Enter `M84 X` or `Y`

### Version of Marlin Firmware

bugfix2.1.x

### Printer model

Voxelab Aquila

### Electronics

Creality CR-Touch

### LCD/Controller

Creality 4.2.7

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

Pronterface

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

robherc, 2025-05-03:
is that a core-kinematics, or delta printer?

ellensp, 2025-05-03:
The creality v427 has 1 enable line for all stepper drivers

It is physically impossible to just disable one stepper driver

This is the expected behavior

closed 2025-05-03 by ellensp

ThomasToka, 2025-05-03:
Shared Pins.

M84 followed by any axis will fail.
M84 disables all axes as they have a shared stop pin.

Same on your board as on those for the E3S1PRO.

classicrocker883, 2025-05-04:
its not a delta. but yeah, you disable one, you disable all for this board unfortunately