MarlinFirmware/Marlin #27833: [BUG] Marlin freezes while homing/ levelling via BL Touch

KLOSTERS, 2025-05-04:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

when executed homing or levelling using BL Touch probe - printer freezes and does not respond to any G1, M114 or M122 command until power or hardware reset.

Behaviour first occurred on Marlin-2.1.2.2 (downloaded on 2025.03.17), when started to test G29: it froze sporadically on homing, on first, second or fifth mesh point.

On Marlin-bugfix-2.1.x (downloaded on 2025.05.04) it freezes at the preceding homing command (G28).

Config files and pronterface console output are appended in the *.zip file:
[20250504_Calibration_Marlin-bugfix-2.1.x.zip](https://github.com/user-attachments/files/20028040/20250504_Calibration_Marlin-bugfix-2.1.x.zip)

### Expected behavior

when G28 is executed:

Bed goes away from the probe (Z-Axis, down)
homing X starts
homing Y starts
extruder moves to bed centre
BL touch probe goes down (deploys)
bed goes towards probe (up) until it touches probe once
BL touch probe goes up (stowes)
**bed moves down (away from probe)**
BL touch probe goes down (deploys)
bed goes towards probe (up) until it touches probe once
BL touch probe goes up (stowes)
bed moves down (away from probe)

### Actual behavior

when G28 is executed:

Bed goes away from the probe (Z-Axis, down)
homing X starts
homing Y starts
extruder moves to bed centre
BL touch probe goes down (deploys)
bed goes towards probe (up) until it touches probe once
BL touch probe goes up (stowes)
**bed (z-axis) does not move anymore, but the motor isn't disabled**
sometimes BL touch probe starts to deploy and stowe after a long delay

### Steps to Reproduce

execute G28 and G29.

### Version of Marlin Firmware

Marlin-bugfix-2.1.x

### Printer model

my own built CORE XY

### Electronics

BIGTREETECH-SKR 3 EZ V1.0, EZ5160 V1.0 drivers, 3D Touch probe.

### LCD/Controller

BIGTREETECH-TFT24-V1.1

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Pronterface

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[BIGTREETECH-TFT24-V1.1_config_Files.zip](https://github.com/user-attachments/files/20028192/BIGTREETECH-TFT24-V1.1_config_Files.zip)

ellensp, 2025-05-04:
Expected is not correct
In Marlin With Z_SAFE_HOMING 
~extruder~probe  moves to bed center
Ie it takes into account your NOZZLE_TO_PROBE_OFFSET

Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.1.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

ellensp, 2025-05-04:
The debugging I now see in the first zip

ellensp, 2025-05-04:
the TMC5160's  

you have
      #define Z_RSENSE          0.256  
     #define E0_RSENSE         0.256
vs 
    #define X_RSENSE		0.075
    #define Y_RSENSE		0.075

so two of these are probably incorrect

KLOSTERS, 2025-05-04:
Hey @ellensp,
thanks for the quick reply and the hint about RSENSE. 
I fixed that, but the behaviour did not really change: the firmware responds now, M114 reports, that Z axis has moved, but it did not move.
as for ridiculously high steps/mm for Z:
 - M6 threaded rods are used for moving bed
 - 1,8 deg/step => 200 Steps/ rotation of the motor
 - with 2,5 translation ratio between Motor and spindle and 1mm per spindle rotation => 500 steps per rotation
 - with 256 microsteps/step and 500 steps per rotation => 128000 steps per unit
 I also measured the calibration values. they work for Z axis.

KLOSTERS, 2025-05-04:
Yes I actually did not know, what the INTERPOLATE did. 
After setting MICROSTEP to 32 the printer finally works. 

Thanks a lot!

closed 2025-05-04 by KLOSTERS