MarlinFirmware/Marlin #27834: After using manual four-corner leveling, the homing  function no longer works properly.

graygit2024hub, 2025-05-05:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

the board is MKS_GEN_L_V21, only x use the UART control, after using manual four-corner leveling, the homing  function no longer works properly, stepper motors stuck at end-stop and until the system auto reset.

### Version of Marlin Firmware

2.1.3-b2

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/user-attachments/files/20032169/Configuration.zip)

ellensp, 2025-05-05:
Configuration_adv.h has LCD_BACKLIGHT_TIMEOUT_MINS set, but no LCD_BACKLIGHT_PIN is defined.

Did you change any pins files. If so please attach them

thinkyhead, 2026-10-08:
Following up for the 2.1.3 release: this one is still waiting on the details requested above. Could you test with the current `bugfix-2.1.x` and post your configs and the results? If we don't hear back in a couple of weeks we'll close it, but it can always be reopened.