MarlinFirmware/Marlin #27842: [BUG] #define LCD_WIDTH 16 doesn't work properly. X & Y data is missing from the screen as well.

a4jp-com, 2025-05-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I'm trying to update the firmware of a Scoovo X9H in Visual Studio Code but can't get the 16x4 character settings working.  X & Y data is also missing from the screen making work really difficult, directly on the printer, when I have to use just an SD card instead of the USB cable.

I set these

#define LCD_WIDTH 16 //not working
#define LCD_HEIGHT 4 //okay

So it should work in theory but it's not.

I can see line 195 of Marlin-bugfix-2.1.x\.pio\build\rambo\firmware.elf has the words "3D Printer Ready" so I guess I can check what is displayed in this file. 

I tried editing words in language_en.h file but couldn't fix a way to fix the length of lines and add the missing X and Y data to the info screen. Please help.

I have a VITEK DISPLAY VC1641-KBW-JA (16x4 character screen). It has a home button and 5 button push controller.

[https://www.vitek.com.tw/character-module-pages-2-0.htm]
VC1641
The controller for this is ST7066U or KS0066.
(Maybe it is HD44780 compatible)
----------------------------------------

I'm happy to use or try any controller or lcd setting you set in the settings though.

The real driver
#define LCD_TYPE ST7066U // Or KS0066 if applicable
#define LCD_INTERFACE 8BIT // Use 4BIT if necessary

From the old Marlin 1  file
#define ULTRA_LCD // lcd works but no controller & 20x4
#define ULTIPANEL //lcd and controller work but 20x4

#define LCD_TYPE_HD44780/*

-----------------------------------------
Here is an example of the old screen options in Marlin 1
of ultralcd_implementation_hitachi_HD44780.h

Possible status screens:
16x2   |0123456789012345|
       |000/000 B000/000|
       |Status line.....|

16x4   |0123456789012345|
       |000/000 B000/000|
       |SD100%    Z000.0|
       |F100%     T--:--|
       |Status line.....|

20x2   |01234567890123456789|
       |T000/000D B000/000D |
       |Status line.........|

20x4   |01234567890123456789|
       |T000/000D B000/000D |
       |X+000.0 Y+000.0 Z+000.0|
       |F100%  SD100% T--:--|
       |Status line.........|

20x4   |01234567890123456789|
       |T000/000D B000/000D |
       |T000/000D     Z000.0|
       |F100%  SD100% T--:--|
       |Status line.........|
*/

------------------------------------------

[image]

### Bug Timeline

Marlin 2

### Expected behavior

If I set 
#define LCD_WIDTH 16
#define LCD_HEIGHT 4
There is enough room to show X, Y & Z fully.

The old screen

16x4   |0123456789012345|
       |000/000 B000/000|
       |SD100%    Z000.0|
       |F100%     T--:--|
       |Status line.....|

but I currently get

16x4   |0123456789012345|
       |_22/0___H_21/0°_|
       |_________Z___0___|
       |F100%___________|
       |Status line.....|

This shows all the info we need.

16x4   |0123456789012345|
       |000/000_B000/000|
       |**X**00.0**Y**00.0**Z**000.0| 
       |F100_C100_T--:--|
       |Status_line.....|

|F100_C100_T--:--|
|F99%_C99%_T--:--|

The X, Y & Z would have a dark background but light characters so it is easy to see.
|**X**99.9**Y**99.9**Z**999.9|

When the numbers get too large for the display the first 999 would have a dark background and last 2 decimal numbers 99 would have a light background. This would easily allow all the characters to be seen.
|**999**99**999**99**999**99 |

[image]

### Actual behavior

16x4 is showing up as 20x4

[image]

### Steps to Reproduce

Just install the firmware and set the screen to 16x4

### Version of Marlin Firmware

Marlin Bugfix-2.1.x

### Printer model

Scoovo X9H

### Electronics

Stock setup

### LCD/Controller

A VITEK DISPLAY VC1641-KBW-JA (16x4 character screen). It has a home button and 5 button push controller | #define LCD_TYPE ST7066U // Or KS0066 if applicable | #define LCD_INTERFACE 8BIT // Use 4BIT if necessary | VITEK DISPLAY VC1641

### Other add-ons

None

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Pronterface

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I included so of the older files for the screen as well just incase they help.

[Configuration-and-some-old-files.zip](https://github.com/user-attachments/files/20067839/Configuration-and-some-old-files.zip)

ellensp, 2025-05-07:
Inverted charters are not part of the hd44780 charter rom.

You can customize only 8 characters.  These are all used.
There is not sufficient customization to generate inverted numerics.

Stock 16x4 lcd is meant to be of format

```
 *  16x4   |000/000 B000/000|
 *         |SD---%  Z 000.00|
 *         |F---%     T--:--|
 *         |0123456789012345|

```

Not having X or Y data is a design limit, not a bug.

ellensp, 2025-05-07:
From the marlin simulator a 16c4 lcd

[image]

With 
#define REPRAP_DISCOUNT_SMART_CONTROLLER
#define LCD_WIDTH 16

a4jp-com, 2025-05-07:
Yes, that's the screen I'm seeing now. SD is missing. And shows up on the 3rd line when printing. I think that bit of the code has a bug and switches to 20x4 instead of 16x4.

After that is fixed, can I ask for a new compact LCD_INFO_SCREEN_STYLE 2 : Compact XYZ Screen to get around the design limit if it's okay then?

Here is my idea.

[image]

This would be super useful for people like me with old machines. I don't want to throw away old working parts if possible.

a4jp-com, 2025-05-07:
Here are photos of the screen now.

[image]
[image]

ellensp, 2025-05-07:
find          " TERN_(SHOW_PROGRESS_PERCENT, setPercentPos(0, 2));"  in Marlin/src/lcd/HD44780/marlinui_HD44780.cpp

Change the 2 to a 1, should fix the sdcard positioning.

a4jp-com, 2025-05-07:
Changing the number to 1 fixed the bug. Thank you so much for that super easy fix. But now if it’s okay, can I leave this case open here as a feature request as I still can see the X & Y values on the screen which makes the printer difficult to use.

Photos after the code update.

[image]
[image]

ellensp, 2025-05-07:
I have created a PR to fix the bug.  https://github.com/MarlinFirmware/Marlin/pull/27844
Updating this to a feature request for a alternative lcd style

ellensp, 2025-05-07:
ok, it's not perfect, but, here is a diff of the code changes.

```diff
diff --git a/Marlin/src/lcd/HD44780/marlinui_HD44780.cpp b/Marlin/src/lcd/HD44780/marlinui_HD44780.cpp
index 265f62ab88..cae20720a1 100644
--- a/Marlin/src/lcd/HD44780/marlinui_HD44780.cpp
+++ b/Marlin/src/lcd/HD44780/marlinui_HD44780.cpp
@@ -900,6 +900,13 @@ void MarlinUI::draw_status_message(const bool blink) {
  *         |F---%     T--:--|
  *         |0123456789012345|
  *
+ *  LCD16_INFO_SCREEN_STYLE 1
+ *
+ *  16x4   |000/000 B000/000|
+ *         |X00.0Y00.0Z00.00|
+ *         |F---%SD---T--:--|
+ *         |0123456789012345|
+ *
  *  20x2   |T000/000° B000/000° |
  *         |01234567890123456789|
  *
@@ -1000,9 +1007,32 @@ void MarlinUI::draw_status_screen() {
 
       #if LCD_WIDTH < 20
 
-        #if HAS_PRINT_PROGRESS
-          TERN_(SHOW_PROGRESS_PERCENT, setPercentPos(0, 2));
-          rotate_progress();
+        #if LCD16_INFO_SCREEN_STYLE
+          lcd_moveto(0, 1);
+          #if HAS_X_AXIS
+            if (LOGICAL_X_POSITION(current_position.x) < 100)
+              _draw_axis_value(X_AXIS, ftostr31rj(LOGICAL_X_POSITION(current_position.x)), blink);
+            else
+              lcd_put_u8str(ftostr41rj(LOGICAL_X_POSITION(current_position.x)));
+          #endif
+          #if HAS_Y_AXIS
+            if (LOGICAL_Y_POSITION(current_position.y) < 100)
+              _draw_axis_value(Y_AXIS, ftostr31rj(LOGICAL_Y_POSITION(current_position.y)), blink);
+            else
+              lcd_put_u8str(ftostr41rj(LOGICAL_Y_POSITION(current_position.y)));
+          #endif
+          #if HAS_Z_AXIS
+            if (LOGICAL_Z_POSITION(current_position.z) < 100)
+              _draw_axis_value(Z_AXIS, ftostr42rj(LOGICAL_Z_POSITION(current_position.z)), blink);
+            else
+              lcd_put_u8str(ftostr52rj(LOGICAL_Z_POSITION(current_position.z)));
+          #endif
+        #else // LCD16_INFO_SCREEN_STYLE
+
+          #if HAS_PRINT_PROGRESS
+            TERN_(SHOW_PROGRESS_PERCENT, setPercentPos(0, 1));
+            rotate_progress();
+          #endif
         #endif
 
       #else // LCD_WIDTH >= 20
@@ -1072,11 +1102,13 @@ void MarlinUI::draw_status_screen() {
 
       #endif // LCD_WIDTH >= 20
 
-      #if HAS_Z_AXIS
-        lcd_moveto(LCD_WIDTH - 8, 1);
-        _draw_axis_value(Z_AXIS, ftostr52sp(LOGICAL_Z_POSITION(current_position.z)), blink);
-        #if HAS_LEVELING && !HAS_HEATED_BED
-          lcd_put_lchar(planner.leveling_active || blink ? '_' : ' ');
+      #if LCD16_INFO_SCREEN_STYLE != 1
+        #if HAS_Z_AXIS
+          lcd_moveto(LCD_WIDTH - 8, 1);
+          _draw_axis_value(Z_AXIS, ftostr52sp(LOGICAL_Z_POSITION(current_position.z)), blink);
+          #if HAS_LEVELING && !HAS_HEATED_BED
+            lcd_put_lchar(planner.leveling_active || blink ? '_' : ' ');
+          #endif
         #endif
       #endif
 
@@ -1090,6 +1122,14 @@ void MarlinUI::draw_status_screen() {
       lcd_put_u8str(i16tostr3rj(feedrate_percentage));
       lcd_put_u8str(F("%"));
 
+      #if LCD16_INFO_SCREEN_STYLE
+        #if HAS_PRINT_PROGRESS
+          TERN_(SHOW_PROGRESS_PERCENT, setPercentPos(5, 2);
+        );
+          rotate_progress();
+        #endif
+      #endif
+
       #if LCD_WIDTH >= 20
 
         #if HAS_PRINT_PROGRESS
```

add #define LCD16_INFO_SCREEN_STYLE 1 to your config

I didn't use LCD_INFO_SCREEN_STYLE, as there is code that stops that being for 16 char displays

Current issues is some strings overwrite others.
for eg the SD percent always has a % on the end, which it over written with the elasped time. E

I would not  want to add this to marlin, as I can just see the string of error reports. "BUG, It prints positions as 101.5130.5 Z 10.10, losing the X and Y"

a4jp-com, 2025-05-07:
I figured out how to add Style 3 in Marlin/src/lcd/HD44780/marlinui_HD44780.cpp. Sorry I didn't see your code here till now. I was working on this all day :/ But at least I set mine up as a MiniXYZ style so people can just turn it on through Configuration.h

First I set Style 3 in Configuration.h and wrote a warning
`/**
 * Info Screen Style (0:Classic, 1:Průša, 2:CNC, 3:MiniXYZ)
 *
 * :[0:'Classic', 1:'Průša', 2:'CNC', 3:'MiniXYZ']
Note: With MiniXYZ on a 16x4 LCD the X, Y & Z letters will be switched with numbers when they get into the hundreds.
This is not a bug it's to get around the character limits of the screen for people that want X, Y & Z data. 
 */`

Then in Marlin/src/lcd/HD44780/marlinui_HD44780.cpp I did this.
First add this code to the top of the file:

`//for line 1270 to remove the plus mark on MiniXYZ
#include <stdlib.h>  // abs()
#include <stdio.h>   // sprintf()
#include <string.h>  // strcat()

char* ftostr41_zeropad(const float &x) {
  static char out[6]; // "000.0"
  const int xx = (int)(x * 10 + 0.5);
  if (xx < 0)
    sprintf(out, "-%02d.%01d", abs(xx) / 10, abs(xx) % 10);
  else
    sprintf(out, "%03d.%01d", xx / 10, xx % 10);
  return out;
}

char* ftostr52_zeropad(const float &x) {
  static char out[7]; // "000.00"
  const int xx = (int)(x * 100 + 0.5);
  if (xx < 0)
    sprintf(out, "-%02d.%02d", abs(xx) / 100, abs(xx) % 100);
  else
    sprintf(out, "%03d.%02d", xx / 100, xx % 100);
  return out;
}

char* format_axis(float value, char axis) {
  static char out[10];
  if (axis == 'Z' && value < 100.0) {
    int zz = (int)(value * 100 + 0.5);
    sprintf(out, "%c%02d.%02d", axis, zz / 100, zz % 100);
  } else if (value < 100.0) {
    int xx = (int)(value * 10 + 0.5);
    sprintf(out, "%c%02d.%01d", axis, xx / 10, xx % 10);
  } else {
    int hundred = (int)(value / 100);
    int remainder = (axis == 'Z') ? (int)((value - (hundred * 100)) * 100 + 0.5) : (int)((value - (hundred * 100)) * 10 + 0.5);
    sprintf(out, "%d%02d.%0*d", hundred, remainder / ((axis == 'Z') ? 100 : 10), (axis == 'Z') ? 2 : 1, remainder % ((axis == 'Z') ? 100 : 10));
  }
  return out;
}
`

Then add Style 3 to the main code in Configuration.h.

`void MarlinUI::draw_status_screen() {

  const bool blink = get_blink();
  lcd_moveto(0, 0);

  #if LCD_INFO_SCREEN_STYLE == 0

    // ========== Line 1 ==========

    #if LCD_WIDTH < 20

      //
[... 416 lines cut ...]
          const uint32_t secs = time_secs % 60;

          // Output feedrate character and value
          lcd_put_lchar(LCD_STR_FEEDRATE[0]); // Single character for "FR"
          lcd_put_u8str(i16tostr3rj(feedrate_percentage)); // e.g., "100"

          // Format time only
          if (days > 0) {
            sprintf(buf, "T%ldd%02ldh%02ld", days, hours, mins);
          } else if (hours >= 1) {
            sprintf(buf, " T%02ldh%02ldm", hours, mins);
          } else {
            sprintf(buf, " T%02ldm%02lds", mins, secs);
          }
        #endif
        lcd_put_u8str(buf);
      #endif
    #endif // LCD_HEIGHT > 3
    
    // ========== Line 4 (Status Message) ==========
    draw_status_message(blink);
}
#endif // LCD_INFO_SCREEN_STYLE`

I did a few test prints with the code as well and it's almost perfect now.

a4jp-com, 2025-05-07:
I made this new MiniXYZ style as I didn't wanna lose Style 0, 1 or 2. 

I also want to edit the default 20x4 screen in the MiniXYZ style as well even though I don't have a printer with that type of screen.
It would just be useful having all the values on screen X00.00 Y00.00 Z00.00 (100.00 100.00 100.00).

a4jp-com, 2025-05-07:
Here is a photo of the MiniXYZ style screen. I even got the little temperature icon working at the beginning of the first line.

[image]

a4jp-com, 2025-05-07:
I'm just tweaking the code now so it looks nicer. I'll put the letter P for percent or print next to the number instead of leaving a large gap. And I'll adjust the switching time between time displays. Maybe 4 seconds between showing the elapsed time and remaining time if the remaining time is turned on in the settings. 2 seconds was a bit too fast.

a4jp-com, 2025-05-08:
I fix everything and got the code working properly. I'll test it and if everything is okay I want to submit it.

```
// ========== Line 3 ==========
    #if LCD_HEIGHT > 3
      char buf[10] = {0}; // Buffer for time only (max 8 chars + null)
      lcd_moveto(0, 2); // Start at column 0 for feedrate

      // Feedrate, SD Progress, and Time
      #if LCD_WIDTH >= 16
        uint32_t time_secs;
        bool is_remaining = false;
        static uint8_t toggle_counter = 0;
        static bool toggle_state = false;

        // Increment counter every call (~1s), toggle every 6 calls (~6s)
        if (++toggle_counter >= 6) {
          toggle_counter = 0;
          toggle_state = !toggle_state;
        }

        #if ENABLED(SDSUPPORT) && HAS_PRINT_PROGRESS
          if (card.isFileOpen()) {
            #if ENABLED(SHOW_ELAPSED_TIME) && ENABLED(SHOW_REMAINING_TIME)
              // Alternate between elapsed and remaining time during print
              is_remaining = toggle_state;
              time_secs = is_remaining ? ui.get_remaining_time() : print_job_timer.duration();
            #elif ENABLED(SHOW_REMAINING_TIME)
              time_secs = ui.get_remaining_time(); // Remaining time only
              is_remaining = true;
            #elif ENABLED(SHOW_ELAPSED_TIME)
              time_secs = print_job_timer.duration(); // Elapsed time only
            #else
              time_secs = print_job_timer.duration(); // Default to elapsed
            #endif
          } else {
            time_secs = print_job_timer.duration(); // Elapsed time post-print
          }

          // Calculate time components
          const uint32_t days = time_secs / 86400;
          const uint32_t hours = (time_secs % 86400) / 3600;
          const uint32_t mins = (time_secs % 3600) / 60;
          const uint32_t secs = time_secs % 60;

          // Get SD progress
          int progress = card.isFileOpen() ? (int)card.percentDone() : 100;

          // Output feedrate at column 0
          lcd_put_lchar(LCD_STR_FEEDRATE[0]); // Single character for "F"
          lcd_put_u8str(i16tostr3rj(feedrate_percentage)); // e.g., "100"

          // Output progress with dynamic column start
          #if ENABLED(SHOW_PROGRESS_PERCENT)
            char prog_buf[5] = {0};
            sprintf(prog_buf, "P%d", progress); // "P100", "P98", "P8"
            int col_start = (progress >= 100) ? 4 : (progress >= 10) ? 5 : 6;
            lcd_moveto(col_start, 2);
            lcd_put_u8str(prog_buf);
          #endif

          // Format time only, starting at column 9
          lcd_moveto(9, 2);
          #if ENABLED(SHOW_PROGRESS_PERCENT)
            if (days > 0) {
              // Format: T2d12h30 or R2d12h30
              sprintf(buf, "%c%ldd%02ldh%02ld", is_remaining ? 'R' : 'T', days, hours, mins);
            } else if (hours >= 1) {
              // Format: T12h30m or R12h30m
              sprintf(buf, "%c%02ldh%02ldm", is_remaining ? 'R' : 'T', hours, mins);
            } else {
              // Format: T08m49s or R08m49s
              sprintf(buf, "%c%02ldm%02lds", is_remaining ? 'R' : 'T', mins, secs);
            }
          #else
            // Without progress, same time formats
            if (days > 0) {
              sprintf(buf, "%c%ldd%02ldh%02ld", is_remaining ? 'R' : 'T', days, hours, mins);
            } else if (hours >= 1) {
              sprintf(buf, "%c%02ldh%02ldm", is_remaining ? 'R' : 'T', hours, mins);
            } else {
              sprintf(buf, "%c%02ldm%02lds", is_remaining ? 'R' : 'T', mins, secs);
            }
          #endif
        #else
          // Fallback if SDSUPPORT is disabled
          time_secs = print_job_timer.duration(); // Elapsed time only
          const uint32_t days = time_secs / 86400;
          const uint32_t hours = (time_secs % 86400) / 3600;
          const uint32_t mins = (time_secs % 3600) / 60;
          const uint32_t secs = time_secs % 60;

          // Output feedrate at column 0
          lcd_put_lchar(LCD_STR_FEEDRATE[0]); // Single character for "F"
          lcd_put_u8str(i16tostr3rj(feedrate_percentage)); // e.g., "100"

          // Format time only at column 9
          lcd_moveto(9, 2);
          if (days > 0) {
            sprintf(buf, "T%ldd%02ldh%02ld", days, hours, mins);
          } else if (hours >= 1) {
            sprintf(buf, "T%02ldh%02ldm", hours, mins);
          } else {
            sprintf(buf, "T%02ldm%02lds", mins, secs);
          }
        #endif
        lcd_put_u8str(buf);
      #endif
    #endif // LCD_HEIGHT > 3
```

How do I submit this now

a4jp-com, 2025-05-09:
Found a small bug. I moved the z-axis to 110mm but the display hid the .00 bit. Gotta fix that next.

[image]

a4jp-com, 2025-06-23:
I've been testing the new style and just have one bug left to fix. When moving the print head, sometimes the numbers are off by a bit.