MarlinFirmware/Marlin #27843: [BUG] Filament runout detection doesn't work after #27812

vovodroid, 2025-05-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

After commit  **Separate endstop and motion sensor thresholds in filament runout detection #27812** runout detection doesn't work with SKR-3 and BTT SFS V2.0.

Before this commit both runout and motion detection used the same value, now motion detection uses `FILAMENT_MOTION_DISTANCE_MM`, but no runout detection, doesn't matter what `FILAMENT_RUNOUT_DISTANCE_MM` is.

 BTT SFS is connected to board according to BTT instruction https://github.com/bigtreetech/smart-filament-detection-module/blob/master/V2.0/Manual/SFS%20V2.0%20User%20Manual_20231123.pdf:

[image]

### Version of Marlin Firmware

bugfix-2.1.x

### Electronics

SKR-3, BTT SFS V2.0

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/user-attachments/files/20069605/Marlin.zip)

thinkyhead, 2025-09-26:
See the notes added to the problematic PR #27812 on how this should be resolved. In particular, the motion sensor should always trigger as soon as its threshold is reached, and this should be treated separately from the other kind of runout. The simplest way to accomplish this would be to set the counter that tracks `FILAMENT_RUNOUT_DISTANCE_MM` to its limit when the motion sensor detects a jam.

thinkyhead, 2025-09-26:
I believe #28074 should fix the issue, but it only addresses the main problem of wrongly waiting for the runout distance in the case of a filament jam.

closed 2025-09-26 by thinkyhead