MarlinFirmware/Marlin #27849: [BUG] Conditionals looking for Configuration_adv.h defines before they are defined

ellensp, 2025-05-08:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

While looking at BACKLASH_GCODE I found an issue where required defines where accidentally undefined.

This lead down a rabbit hole of finding various Configuration_adv.h tests in Conditionals before Configuration_adv.h was included.

So a write a test script to read in all  Configuration_adv.h defines and check if they are in Conditionals 1,2 and 3 

Results

Keyword 'ULTIPANEL_FEEDMULTIPLY' from Configuration_adv.h found in Marlin/src/inc/Conditionals-2-LCD.h.
Keyword 'BOOT_MARLIN_LOGO_SMALL' from Configuration_adv.h found in Marlin/src/inc/Conditionals-2-LCD.h.
Keyword 'BACKLASH_GCODE' from Configuration_adv.h found in Marlin/src/inc/Conditionals-3-etc.h

 

### Bug Timeline

some time now... 

### Expected behavior

tests that check for defines, but be run after the define is set

### Actual behavior

A small number of tests are run and action taken before the define is set

### Version of Marlin Firmware

bugfix-2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2025-05-09:
Pretty sure I ran into this over in https://github.com/MarlinFirmware/Marlin/issues/27807#issuecomment-2816864518

classicrocker883, 2025-05-11:
I am trying to recreate this, such as enabling `ULTIPANEL_FEEDMULTIPLY` in  **Configuration_adv.h**.  
it seems there is a safeguard: `ULTIPANEL_FEEDMULTIPLY` is under `HAS_MANUAL_MOVE_MENU` and `IS_ULTIPANEL`

however, if this wasn't known already, anything in **inc/Conditionals-[...].h** takes its definitions from **Configuration.h**, so anything enabled under **Configuration_adv.h** will be ignored.

It may break things to have those files enable stuff for only two things which more or less go ignored.

so I would just omit those (`ULTIPANEL_FEEDMULTIPLY`, `BACKLASH_GCODE`) from **Conditionals-2/3-.h**

either that, or somehow move them from **Configuration_adv.h** to **Configuration.h**

PR #27863 (2025-05-19): 🔧 Fix speed/flow edit options
Fix conditionals related to speed / flow edit options and probe feedrate.

Fixes #27849
commit ade6dbf0 (2025-05-19): 🔧 Fix Speed/Flow edit options (#27863)

Fixes #27849
closed 2025-05-19 by thinkyhead