MarlinFirmware/Marlin #27885: [BUG] Sensorless homing does not work with 2240s & potential reboots

thisiskeithb, 2025-05-26:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Testing was done on a BIQU BX with 2240s installed on X, Y, Z1, Z2, and E0 with sensorless homing.

---

I can move each axis around, but sensorless homing does not work and drivers shut down. There are also still compile issues when enabling `TMC_DEBUG`.

### Details

I tried a standard `G28` and there's an slight attempt to home X & Y, but they instantly trigger (or so I thought) and I can no longer command X or Y to move until a power cycle.

I tried enabling `TMC_DEBUG` to collect more info, but it won't compile to due various references to TSTEP. I had to disable the following in `tmc_util.cpp` to get `TMC_DEBUG` to compile:

<details><summary>Click me for the diff!</summary>
<p>

```diff
diff --git a/Marlin/src/feature/tmc_util.cpp b/Marlin/src/feature/tmc_util.cpp
index fb17e56..e7f0414 100644
--- a/Marlin/src/feature/tmc_util.cpp
+++ b/Marlin/src/feature/tmc_util.cpp
@@ -715,13 +715,13 @@
   #endif
 
   template <typename TMC>
-  void print_tstep(TMC &st) {
-    const uint32_t tstep_value = st.TSTEP();
-    if (tstep_value != 0xFFFFF)
-      SERIAL_ECHO(tstep_value);
-    else
-      SERIAL_ECHOPGM("max");
-  }
+  // void print_tstep(TMC &st) {
+  //   const uint32_t tstep_value = st.TSTEP();
+  //   if (tstep_value != 0xFFFFF)
+  //     SERIAL_ECHO(tstep_value);
+  //   else
+  //     SERIAL_ECHOPGM("max");
+  // }
   void print_tstep(TMC2660Stepper &st) { }
 
   template <typename TMC>
@@ -749,7 +749,7 @@
       case TMC_CS_ACTUAL: print_cs_actual(st); break;
       case TMC_VSENSE: print_vsense(st); break;
       case TMC_MICROSTEPS: SERIAL_ECHO(st.microsteps()); break;
-      case TMC_TSTEP: print_tstep(st); break;
+      //case TMC_TSTEP: print_tstep(st); break;
       #if ENABLED(HYBRID_THRESHOLD)
         case TMC_TPWMTHRS: SERIAL_ECHO(uint32_t(st.TPWMTHRS())); break;
         case TMC_TPWMTHRS_MMS: {
@@ -1006,7 +1006,7 @@
         PRINT_TMC_REGISTER(GSTAT);
         PRINT_TMC_REGISTER(IOIN);
         PRINT_TMC_REGISTER(TPOWERDOWN);
-        PRINT_TMC_REGISTER(TSTEP);
+        //PRINT_TMC_REGISTER(TSTEP);
         PRINT_TMC_REGISTER(TPWMTHRS);
         PRINT_TMC_REGISTER(CHOPCONF);
         PRINT_TMC_REGISTER(PWMCONF);
```

</p>
</details>

With `TMC_DEBUG` enabled, I found that the following differences between `M122` reports before & after homing:

Before homing:
```
        X    Y    Z    Z2    E
Enabled        false    false    true    true    false
[...snip...]
Driver registers:
        X    0x81:00:40:00
        Y    0x81:00:40:00
        Z    0x81:0F:40:00
        Z2    0x81:0F:40:00
        E    0x81:00:40:00
```

After homing:
```
        X    Y    Z    Z2    E
Enabled        true    true    true    true    false
[...snip...]
Driver registers:
        X    0x80:0F:50:22
        Y    0x80:0F:50:24
        Z    0x80:1F:40:7C
        Z2    0x80:1F:40:10
        E    0x81:00:40:00
```

@dbuezas found that according to the datasheet, this is a driver shutdown due to overheating. That is not really the case since drivers are cool to the touch and they are triggering instantly while homing, even at the least sensitive StallGuard value.

---

Based on reports in the #trinamic channel on our Discord, you can use 5130 as a substitute for 2240, but I didn't have 100% success with that either. Drivers no longer shutdown, but they do not trigger while homing, even at the most sensitive StallGuard setting (opposite of above since hit state has to be flipped per the sanity checks).

There's also a report of a printer rebooting with fixes from https://github.com/MarlinFirmware/Marlin/pull/27880, but I have not experienced that issue.

Another interesting find is that 2240s are *a lot quieter* when configured as 5130s vs. 2240s.

### Bug Timeline

Since 2240 suppot was added in https://github.com/MarlinFirmware/Marlin/pull/25974

### Expected behavior

Sensorless homing and drivers work

### Actual behavior

Sensorless homing fails and causes drivers to shutdown until a full power cycle

### Steps to Reproduce

1. Install 2240 drivers into BX, including three additional jumpers per driver for SPI mode
2. Update config to support 2240s
3. Build & flash current `bugfix-2.1.x` to BX
4. Attempt to home all via LCD or `G28`

### Version of Marlin Firmware

bugfix-2.1.x (fde0eaf)

### Printer model

BIQU BX

### Electronics

Stock electronics except for 2240 drivers

### LCD/Controller

Stock TFT

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Stock BX config: https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/BIQU/BX

Change all `*_DRIVER_TYPE`s to `2240` and enable `TMC_USE_SW_SPI` or use the attached modified config: [Configurations-config-examples-BIQU-BX.zip](https://github.com/user-attachments/files/20446655/Configurations-config-examples-BIQU-BX.zip)

Don't forget to install three additional jumpers under each driver for SPI mode.

dbuezas, 2025-05-28:
I think I read the bits in the wrong order and there were no fault bits actually.

thisiskeithb, 2025-06-14:
Reopening since this still isn't fixed.