MarlinFirmware/Marlin #27898: [FR] Rotate the print by 90, 180, and 270 degrees on square beds.

pi-squared-studio, 2025-05-31:
### Describe the feature you want

It would be nice to be able to specify the angle of rotation on the bed when selecting a print file.
I'll explain what it's for: I many print with various artistic plastics. In most cases, these are two- and three-color silk PLA plastics. When installed in an extruder, they are usually placed in only one position. For example, if I use red-yellow-green plastic, then I will always see the red side. And this is correct, since the entire whole print time the filament should not change its position. Otherwise, the print will turn out to be defective, or so-called twists of the thread.
It is very difficult to select and change such a position mechanically, and it is rigidly linked with the kinematics of thread feeding.

In such cases, I have artistic ideas to choose the "front" color. Let's say that the red color is not the face, but the back of the model. Or make a batch of such identical models so that they differ in gradient from each other.

I think that such a very simple function will be useful for many! In fact, you just need to replace the X coordinates with Y during the printing process.

### Additional context

If the above really turns out to be useful, then you can also think about implementing a mirror reflection function... It will be useful for cases when there is a g-code for the right model, but you need to print the same left one without much manipulation.

ellensp, 2025-05-31:
"you just need to replace the X coordinates with Y during the printing process"  
No that produces a reflection across the line y = x. It is not a rotation.

Any text on the top surfaces would be backwards 

eg yellow is starting object, blue is same object with x and y swapped

[image]

pi-squared-studio, 2025-05-31:
> No that produces a reflection across the line y = x. It is not a rotation.

Of course, I'm exaggerating, and I said that to simplify the thought. )) The real algorithm is a bit detailed, and takes into account negative values. But it doesn't complicate the code much.
That is, to rotate clockwise by 90 degrees, it will be enough to rewrite the coordinates X1 = Y, Y1 = -X...

thisiskeithb, 2025-05-31:
This is a perfect task for a python script: [rotate_gcode.py.zip](https://github.com/user-attachments/files/20533870/rotate_gcode.py.zip)

Usage: `python3 rotate_gcode.py [--center-x CENTER_X] [--center-y CENTER_Y] input_file output_file rotation`

Example: `python3 rotate_gcode.py --center-x 120 --center-y 120 3dbenchy.gcode 3dbenchy_rotated.gcode 180`

| Stock Benchy | Rotated Benchy  |
|---|---|
| [image] | [image] |

pi-squared-studio, 2025-05-31:
> his is a perfect task for a python script: [rotate_gcode.py.zip](https://github.com/user-attachments/files/20533870/rotate_gcode.py.zip)

You misunderstood the idea a bit. I know how to rotate models using a slicer or script. But here the idea is to be able to quickly change the phase, cough, or orientation of the filament without removing it from the extruder. This is, of course, a narrowly focused improvement, but getting a mirror image without a slicer or script would be useful to many. Imagine that you need to make a mod for your car, and you only have the g-code of only the right exterior mirror body...)))

robherc, 2025-06-02:
Unfortunately no, it wouldn't be "useful to many," simply because there are already too many better places and ways to accomplish this task. Unfortunately, it would be adding a tiny amount of functionality for extremely few, and at the cost of bloating Marlin with features that >99% of users would never use, but many would be forced to remove in order to get a compiled Marlin Firmware image to fit on their resource-constrained (and already overloaded) 8-bit boards. :-/

DerAndere1, 2025-06-06:
For rotation of workspace coordinate systems, LinuxCNC has the G10 L2 Rn command where n is the degree of rotation. If someone wants do addARD this feature, please do it that way and do not invent a new command. A more creative solution would be to install a horizontal rotary table and configure it as a 4th axis (AXIS4_ROTATES, AXIS4_NAME 'C'). In my fork i have support for these kinematics.