MarlinFirmware/Marlin #27902: [BUG] `NONLINEAR_EXTUSION`

classicrocker883, 2025-06-01:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

```yml
Marlin/src/module/stepper.cpp: In static member function 'static void Stepper::calc_nonlinear_e(uint32_t)':
Marlin/src/module/stepper.cpp:2240:38: error: 'uint32_t nonlinear_t::scale_q24' is protected within this context
 2240 |     const uint32_t velocity_q24 = ne.scale_q24 * step_rate; // Scale step_rate first so all intermediate values stay in range of 8.24 fixed point math
      |                                      ^~~~~~~~~
In file included from Marlin/src/module/stepper.cpp:86:
Marlin/src/module/stepper.h:316:18: note: declared protected here
  316 |         uint32_t scale_q24;
      |                  ^~~~~~~~~
Marlin/src/module/stepper.cpp:2244:60: error: 'int32_t nonlinear_t::edividend' is protected within this context
 2244 |     advance_dividend.e = (uint64_t(ne.q24.C + vd_q24) * ne.edividend) >> 24;
      |                                                            ^~~~~~~~~
In file included from Marlin/src/module/stepper.cpp:86:
Marlin/src/module/stepper.h:315:17: note: declared protected here
  315 |         int32_t edividend;
      |                 ^~~~~~~~~
Marlin/src/module/stepper.cpp: In static member function 'static hal_timer_t Stepper::block_phase_isr()':
Marlin/src/module/stepper.cpp:2829:12: error: 'int32_t nonlinear_t::edividend' is protected within this context
 2829 |         ne.edividend = advance_dividend.e;
      |            ^~~~~~~~~
In file included from Marlin/src/module/stepper.cpp:86:
Marlin/src/module/stepper.h:315:17: note: declared protected here
  315 |         int32_t edividend;
      |                 ^~~~~~~~~
Marlin/src/module/stepper.cpp:2830:39: error: 'int32_t nonlinear_t::edividend' is protected within this context
 2830 |         const float scale = (float(ne.edividend) / advance_divisor) * planner.mm_per_step[E_AXIS_N(current_block->extruder)];
      |                                       ^~~~~~~~~
In file included from Marlin/src/module/stepper.cpp:86:
Marlin/src/module/stepper.h:315:17: note: declared protected here
  315 |         int32_t edividend;
      |                 ^~~~~~~~~
Marlin/src/module/stepper.cpp:2831:12: error: 'uint32_t nonlinear_t::scale_q24' is protected within this context
 2831 |         ne.scale_q24 = _BV32(24) * scale;
      |            ^~~~~~~~~
In file included from Marlin/src/module/stepper.cpp:86:
Marlin/src/module/stepper.h:316:18: note: declared protected here
  316 |         uint32_t scale_q24;
```

### Bug Timeline

✨ NONLINEAR_EXTRUSION_DEFAULT_ON (#27819) a6bfdf3 

### Expected behavior

No error

### Actual behavior

Error

### Steps to Reproduce

1. Enable `NONLINEAR_EXTRUSION`

### Version of Marlin Firmware

bugfix-2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2025-06-02:
One way forward is
```diff
diff --git a/Marlin/src/module/stepper.cpp b/Marlin/src/module/stepper.cpp
index fd3d3bd800..1e4a508732 100644
--- a/Marlin/src/module/stepper.cpp
+++ b/Marlin/src/module/stepper.cpp
@@ -2237,11 +2237,11 @@ hal_timer_t Stepper::calc_timer_interval(uint32_t step_rate) {

 #if NONLINEAR_EXTRUSION_Q24
   void Stepper::calc_nonlinear_e(const uint32_t step_rate) {
-    const uint32_t velocity_q24 = ne.scale_q24 * step_rate; // Scale step_rate first so all intermediate values stay in range of 8.24 fixed point math
+    const uint32_t velocity_q24 = ne.get_edividend() * step_rate; // Scale step_rate first so all intermediate values stay in range of 8.24 fixed point math
     int32_t vd_q24 = ((((int64_t(ne.q24.A) * velocity_q24) >> 24) * velocity_q24) >> 24) + ((int64_t(ne.q24.B) * velocity_q24) >> 24);
     NOLESS(vd_q24, 0);

-    advance_dividend.e = (uint64_t(ne.q24.C + vd_q24) * ne.edividend) >> 24;
+    advance_dividend.e = (uint64_t(ne.q24.C + vd_q24) * ne.get_edividend()) >> 24;
   }
 #endif

@@ -2826,9 +2826,9 @@ hal_timer_t Stepper::block_phase_isr() {

       // Calculate Nonlinear Extrusion fixed-point quotients
       #if NONLINEAR_EXTRUSION_Q24
-        ne.edividend = advance_dividend.e;
-        const float scale = (float(ne.edividend) / advance_divisor) * planner.mm_per_step[E_AXIS_N(current_block->extruder)];
-        ne.scale_q24 = _BV32(24) * scale;
+        ne.set_edividend(advance_dividend.e);
+        const float scale = (float(ne.get_edividend()) / advance_divisor) * planner.mm_per_step[E_AXIS_N(current_block->extruder)];
+        ne.set_scale_q24(_BV32(24) * scale);
         if (ne.settings.enabled && current_block->direction_bits.e && ANY_AXIS_MOVES(current_block)) {
           ne.q24.A = _BV32(24) * ne.settings.coeff.A;
           ne.q24.B = _BV32(24) * ne.settings.coeff.B;
diff --git a/Marlin/src/module/stepper.h b/Marlin/src/module/stepper.h
index 90cbedc9fc..c5ceae60ee 100644
--- a/Marlin/src/module/stepper.h
+++ b/Marlin/src/module/stepper.h
@@ -314,6 +314,11 @@ constexpr ena_mask_t enable_overlap[] = {
       protected:
         int32_t edividend;
         uint32_t scale_q24;
+      public:
+        void set_edividend(int32_t value) { edividend = value; }
+        int32_t get_edividend() const { return edividend; }
+        void set_scale_q24(uint32_t value) { scale_q24 = value; }
+        uint32_t get_scale_q24() const { return scale_q24; }
     #endif
   } nonlinear_t;

```

But I have no idea how @vovodroid might want to do this

vovodroid, 2025-06-02:
Actually it seems to be related to 119e77b4c6d192bde8c104753bfaf9175e334b8a