MarlinFirmware/Marlin #27918: [BUG] SEGMENT_LEVELED_MOVES causes choppy movement at high speeds

rondlh, 2025-06-14:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I found that my X-axis moves are quite choppy at fast speeds, starting at about 100mm/s.
I have 160 steps per mm for X, and 80 for Y. The Y axis doesn't show the same issue, or perhaps it does at much higher speeds.
The movement gets smooth again when I disable Bed Leveling (M420 S0).
I found that disabling `SEGMENT_LEVELED_MOVES` solves the issue.
Increasing the `LEVELED_SEGMENT_LENGTH` parameter to 50 reduces the issue, but didn't completely solve it.

### Bug Timeline

Unknown

### Expected behavior

Smooth fast movements

### Actual behavior

Choppy fast movements, especially on the axis with a high steps per mm (160 for X-axis).

### Steps to Reproduce

1. Set the X-steps per unit to 160 steps
2. Home (G28)
3. Move X fast (G0X400F9000/G0X0)
4. G29 (probe bed, enable bed leveling)
5. Move X fast (G0X400F9000/G0X0)
6. Disable Bed leveling M420 S0
7. Move X fast (G0X400F9000/G0X0)

### Version of Marlin Firmware

Latest bugfix (2025-06-14)

### Printer model

Custom

### Electronics

Motherboard BTT_SKR_V3_0, TMC5160 steppers for X and Y

### LCD/Controller

BTT TFT35 V3.0 on serial port

### Other add-ons

BLTouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

The X and Y steppers are all 42mm 1.8 degree steppers, height = 60mm.
Z uses a rod of 2mm/turn, which gives 1600 steps per mm for the 42mm 1.8 degree steppers. I reduced this to 100 for testing, but this did not solve the issue.

The zip file contains a small video, the movement stutter is not clearly visible, but the sound gives it away.
The stepper does not lose steps, and prints seem fine, but this cannot be good of course. 
[Axis Stutter.zip](https://github.com/user-attachments/files/20738305/Axis.Stutter.zip)

Here my minimal configuration that shows the issue.
[Configuration_adv.zip](https://github.com/user-attachments/files/20738309/Configuration_adv.zip).

thisiskeithb, 2025-06-14:
This sounds quite similar to https://github.com/MarlinFirmware/Marlin/issues/19217

rondlh, 2025-06-14:
> This sounds quite similar to [#19217](https://github.com/MarlinFirmware/Marlin/issues/19217)

Yes, indeed, this looks very similar. Let me know if there is anything I should test.
I reduced the steps per unit from 1600 to 100 for a test, but that didn't solve the issue, so I do not expect Z-acceleration is the issue. I don't have CLASSIC_JERK enabled.

UPDATE: I added some leveling data above. Unevenness is about 400um over a the 400x400mm bed.

Jegeva, 2025-06-19:
So, i found this bug becuase i also had stuttering (like VERY bad, it was creating holes in the print on my customized ender3 with a Marlin 2.1.2.5).
I totally cleaned the build and tried changing to disable SEGMENT_LEVELED_MOVES and DEFAULT_MAX_ACCELERATION. but it didnt work.

BUT, since the build was clean i saw these warnings :

`Compiling .pio/build/STM32F103RE_creality/src/src/gcode/control/M17_M18_M84.cpp.o
Marlin/src/feature/bedlevel/abl/bbl.cpp: In static member function 'static void LevelingBilinear::line_to_destination(const_feedRate_t, uint16_t, uint16_t)':
Marlin/src/feature/bedlevel/abl/bbl.cpp:366:47: warning: narrowing conversion of '((current_position.XYZEval<float>::<anonymous>.XYZEval<float>::<unnamed union>::<anonymous>.XYZEval<float>::<unnamed union>::<unnamed struct>::x - LevelingBilinear::grid_start.XYval<float>::<anonymous>.XYval<float>::<unnamed union>::<anonymous>.XYval<float>::<unnamed union>::<unnamed struct>::x) * LevelingBilinear::grid_factor.XYval<float>::<anonymous>.XYval<float>::<unnamed union>::<anonymous>.XYval<float>::<unnamed union>::<unnamed struct>::x)' from 'float' to 'short int' [-Wnarrowing]
366 |   #define CELL_INDEX(A,V) ((V - grid_start.A) * ABL_BG_FACTOR(A))
|                           ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
Marlin/src/feature/bedlevel/abl/bbl.cpp:374:19: note: in expansion of macro 'CELL_INDEX'
374 |     xy_int_t c1 { CELL_INDEX(x, current_position.x), CELL_INDEX(y, current_position.y) },
|                   ^~~~~~~~~~
Marlin/src/feature/bedlevel/abl/bbl.cpp:366:47: warning: narrowing conversion of '((current_position.XYZEval<float>::<anonymous>.XYZEval<float>::<unnamed union>::<anonymous>.XYZEval<float>::<unnamed union>::<unnamed struct>::y - LevelingBilinear::grid_start.XYval<float>::<anonymous>.XYval<float>::<unnamed union>::<anonymous>.XYval<float>::<unnamed union>::<unnamed struct>::y) * LevelingBilinear::grid_factor.XYval<float>::<anonymous>.XYval<float>::<unnamed union>::<anonymous>.XYval<float>::<unnamed union>::<unnamed struct>::y)' from 'float' to 'short int' [-Wnarrowing]
366 |   #define CELL_INDEX(A,V) ((V - grid_start.A) * ABL_BG_FACTOR(A))
|                           ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
Marlin/src/feature/bedlevel/abl/bbl.cpp:374:54: note: in expansion of macro 'CELL_INDEX'
374 |     xy_int_t c1 { CELL_INDEX(x, current_position.x), CELL_INDEX(y, current_position.y) },
|                                                      ^~~~~~~~~~
Marlin/src/feature/bedlevel/abl/bbl.cpp:366:47: warning: narrowing conversion of '((destination.XYZEval<float>::<anonymous>.XYZEval<float>::<unnamed union>::<anonymous>.XYZEval<float>::<unnamed union>::<unnamed struct>::x - LevelingBilinear::grid_start.XYval<float>::<anonymous>.XYval<float>::<unnamed union>::<anonymous>.XYval<float>::<unnamed union>::<unnamed struct>::x) * LevelingBilinear::grid_factor.XYval<float>::<anonymous>.XYval<float>::<unnamed union>::<anonymous>.XYval<float>::<unnamed union>::<unnamed struct>::x)' from 'float' to 'short int' [-Wnarrowing]
366 |   #define CELL_INDEX(A,V) ((V - grid_start.A) * ABL_BG_FACTOR(A))
|                           ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
Marlin/src/feature/bedlevel/abl/bbl.cpp:375:19: note: in expansion of macro 'CELL_INDEX'
375 |              c2 { CELL_INDEX(x, destination.x), CELL_INDEX(y, destination.y) };
|                   ^~~~~~~~~~
Marlin/src/feature/bedlevel/abl/bbl.cpp:366:47: warning: narrowing conversion of '((destination.XYZEval<float>::<anonymous>.XYZEval<float>::<unnamed union>::<anonymous>.XYZEval<float>::<unnamed union>::<unnamed struct>::y - LevelingBilinear::grid_start.XYval<float>::<anonymous>.XYval<float>::<unnamed union>::<anonymous>.XYval<float>::<unnamed union>::<unnamed struct>::y) * LevelingBilinear::grid_factor.XYval<float>::<anonymous>.XYval<float>::<unnamed union>::<anonymous>.XYval<float>::<unnamed union>::<unnamed struct>::y)' from 'float' to 'short int' [-Wnarrowing]
366 |   #define CELL_INDEX(A,V) ((V - grid_start.A) * ABL_BG_FACTOR(A))
|                           ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
Marlin/src/feature/bedlevel/abl/bbl.cpp:375:49: note: in expansion of macro 'CELL_INDEX'
375 |              c2 { CELL_INDEX(x, destination.x), CELL_INDEX(y, destination.y) };
|                                                 ^~~~~~~~~~
Marlin/src/feature/bedlevel/abl/bbl.cpp:392:30: warning: narrowing conversion of '_MAX<short int, short int>(c1.XYval<short int>::<anonymous>.XYval<short int>::<unnamed union>::<anonymous>.XYval<short int>::<unnamed union>::<unnamed struct>::x, c2.XYval<short int>::<anonymous>.XYval<short int>::<unnamed union>::<anonymous>.XYval<short int>::<unnamed union>::<unnamed struct>::x)' from 'int' to 'signed char' [-Wnarrowing]
392 |     const xy_int8_t gc { _MAX(c1.x, c2.x), _MAX(c1.y, c2.y) };
|                          ~~~~^~~~~~~~~~~~
Marlin/src/feature/bedlevel/abl/bbl.cpp:392:48: warning: narrowing conversion of '_MAX<short int, short int>(c1.XYval<short int>::<anonymous>.XYval<short int>::<unnamed union>::<anonymous>.XYval<short int>::<unnamed union>::<unnamed struct>::y, c2.XYval<short int>::<anonymous>.XYval<short int>::<unnamed union>::<anonymous>.XYval<short int>::<unnamed union>::<unnamed struct>::y)' from 'int' to 'signed char' [-Wnarrowing]
392 |     const xy_int8_t gc { _MAX(c1.x, c2.x), _MAX(c1.y, c2.y) };
|                                            ~~~~^~~~~~~~~~~~`

So i moved to UBL with 4 point per dimension  (i was using bilinear with 5 point by dimension) and they vanished.
I am currently running a test print but it seems it vanished for me.

Maybe it rings a bell with your side, maybe worth trying to check this change and/or change the mesh levelling model ?

Worth a check, Hope this helps.

rondlh, 2025-08-01:
I had a closer look at the code that is causing trouble for me, it's in planner.cpp and planner.h
It's this function limit_value_by_axis_maximum:

```
      FORCE_INLINE static float limit_value_by_axis_maximum(const_float_t max_value, xyze_float_t &unit_vec) {
        float limit_value = max_value;
        LOOP_LOGICAL_AXES(idx) {
          if (unit_vec[idx]) {
            if (limit_value * ABS(unit_vec[idx]) > settings.max_acceleration_mm_per_s2[idx]) {
              limit_value = ABS(settings.max_acceleration_mm_per_s2[idx] / unit_vec[idx]);
            }
          }
        }
        return limit_value;
      }
```
max_value seems to be MAX(X-acceleration, Y-acceleration, Z-acceleration).
The unit_vec is the difference between the current block motion vector and the previous block motion vector, normalized to a length of 1. 

Let's assume an long X-movement with bed leveling.
The move will be chopped up into small section of a defined size (LEVELED_SEGMENT_LENGTH), but actually the X-movement per segment is not the same because the motion needs to end on a discrete (micro) step number, so practically I can see that the move is 802 X-steps, then 803 X-steps, then 802...etc. The Z-movement is very small, typically below <32 microsteps (16microsteps = 1 full step). As a result the function above will limit the acceleration of the motion block based on the Z-acceleration. Actually there is not much acceleration ongoing, but because of the discretisation noise (802 803 802 steps) is seems there is acceleration and deceleration, which is limited by the defined Z-acceleration, which causes the choppy motion at high speeds. 

**My hacked solutions:**
1. If I completely disable `limit_value_by_axis_maximum` then the problem is gone and nothing seems to go wrong. Z-movements work fine and I can see that there is proper acceleration and deceleration.
2. Another way to get ride of the problem is to not let the Z-axis be the acceleration limiter if the number of Z-steps is less than 2 full steps.

kennovo, 2026-04-20:
Thanks a lot for the testing! It's an interesting problem but I have no workflow for doing tests like that myself.

A sane motion code shouldn't produce this kind of stutter at all, so it's probably premature to try to push this into Marlin. I don't know for sure what's going on right now. Again, by my reading of the algorithm, letting the Z-axis be the acceleration limiter for small Z-displacements should not be a problem if everything works as it's mathematically intended. But I could easily be wrong; the solution might be what you suggested on Aug 1 after all. Though if that's really it, I'd be a bit worried about machines with different mechanics. For example, it would not be appropriate for Creality's CoreXZ offerings, and there are far more exotic contraptions out there. The code seems to be deliberately written to cover every possible case and we'd be degrading that by giving the Z axis preferential treatment.

So my hope is still on the avenue I brought up, which is not entirely exhausted yet. I found these lines [further down in planner.cpp](https://github.com/CrealityOfficial/Ender-3V3-SE/blob/e7b2bad8ff23b3b3723ac6875ed02561a90a8f2b/Marlin/src/module/planner.cpp#L2473) :
`// Fast acos(-t) approximation (max. error +-0.033rad = 1.89°)`
`// NOTE: junction_theta bottoms out at 0.033 which avoids divide by 0.`
```
            const float limit_sqr = (block->millimeters * junction_acceleration) / junction_theta;
            NOMORE(vmax_junction_sqr, limit_sqr);
```
So that's another (similar) formula where the exact same problem occurs: the angle occurs in both junction_acceleration and junction_theta, but the latter is too high (orders of magnitude too high for small angles) so that the speed at the junction between two blocks is artificially limited. If I increase our thresholds further so that they correspond to 0.039rad = 2.2° (not 0.033 because that's the minimum of the approximate formula; I took the first intersection point between the approximate and exact acos), we get:
- in planner.h function normalize_junction_vector(), my earlier `2.0e-6f` would need to be changed to`0.0016f`. Careful though; depending on the units, this might break [the normalization of unit_vec](https://github.com/MarlinFirmware/Marlin/blob/bb2a54284b1d27f52387f3950e52d4bc25931793/Marlin/src/module/planner.cpp#L2542). To play it safe, I recommend creating a separate function normalize_displacement_vector() with the *original* contents of normalize_junction_vector(), and use _that_ function for unit_vec ! (Gee, If it goes on like this, I'll have to make a fork anyway.)
- in planner.h function limit_value_by_axis_maximum(), change my earlier `1.5e-3f` to `0.04f`

Warfare360, 2026-10-04:
I folks , any update regarding current situation ? @kennovo have you eventually been able to fix the root cause of those infamous stutters ^^

kennovo, 2026-10-07:
The challenge (I'm sure it's well-known to anyone working on these things) is that the 3D model goes through several stages. The slicer can predispose the printer to stutter (for instance by generating too many small segments or by setting parameters that confuse the firmware). Within Marlin, I've been addressing a narrow scenario where (the JD code in) **the motion planner module** generates bad output that makes **the stepper controller module** misbehave. But it appears there's more to it, which can only be addressed through better insight in the latter module. To complicate things, stutters could also happen if the calculations in the microcontroller lag behind real time (which I didn't improve by adding a few floating point operations), and then there's [wildcard stuff ](https://github.com/navaismo/Marlin_bugfix_2.1_E3V3SE/issues/72#issuecomment-5318562268)... In short, this needs some quality time to move ahead and I personally can't make any promises on a time line.

Edit: but thanks for asking - it helps prevent it from slipping off the radar entirely.

closed 2026-10-08 by thinkyhead