MarlinFirmware/Marlin #27923: [BUG] (bug summary) POWERING OFF by M81 or GUI problem

gorlov-ai, 2025-06-16:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I came across the bug in the code that I found when tried to turn off the power from the marlin menu (GUI).
The essence of the bug is this: if I try to turn off the printer via the GUI, nothing happens, but if I enter the M81 command through the terminal, everything works fine.
And here's the reason for the bug on the screenshots attached. If you're interested, I can explain.
If I comment "!ui.on_status_screen() ||", everything works fine.

[Marlin cfgs.zip](https://github.com/user-attachments/files/20760853/Marlin.cfgs.zip)

[image]

[image]

[image]

### Bug Timeline

New bug

### Expected behavior

If I activate POWER OFF feature from marlin GUI then printer must shutting down after specified time (in relation to setting it in Configuration.h).

### Actual behavior

If I activate POWER OFF feature from marlin GUI, nothing happens, and printer continue idle.
If I activate POWER OFF feature M81 from command line, then printer was shutting down after specified time.

### Steps to Reproduce

1. Activate power control in "Configuration.h" #define PSU_CONTROL and specify #define PS_ON_PIN.
2. Try to shutting down printer from command line by M81.
3. Try to shutting down printer from marlin GUI.

### Version of Marlin Firmware

"2.0.9.2"

### Printer model

ZENIT

### Electronics

mks_tinybee

### LCD/Controller

MKS_MINI_12864_V3

### Bed Leveling

None

### Your Slicer

IdeaMaker

### Host Software

SD Card (headless)

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin cfgs.zip](https://github.com/user-attachments/files/20760857/Marlin.cfgs.zip)

thisiskeithb, 2025-06-17:
> Version of Marlin Firmware
> "2.0.9.2"

Your attached configs are also `2.0.9.2`, released Oct 3, 2021. This is why we ask you to try the latest `bugfix-2.1.x` since this bug has likely been fixed.

Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

gorlov-ai, 2025-06-17:
Hello.
I'm try to test last version "bugfix-2.1.x" with some base configs for my HW. It seems like the shutdown function works properly. There were some problems with macros during compilation, but I'll deal with that.
Thank you for your help. I'll probably inform the developers from "makerbase-mks" that it's time to update the links to the new branches.

closed 2025-06-17 by thisiskeithb

gorlov-ai, 2025-06-18:
Sorry, but after detailed configuration, this problem returned.
[Marlin cfg.zip](https://github.com/user-attachments/files/20790811/Marlin.cfg.zip)

In addition, there are new bugs when moving at high speeds.