MarlinFirmware/Marlin #27929: [BUG] Preheat menu targets active extruder not selected hotend

RainMotorsports, 2025-06-19:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Temperature Preheat PLA or ABS menu will only target the active extruder regardless of END 1 or END 2 selected. T0/T1 command sent affects which hotend will preheat with either END 1 or END 2. Filament Load Preheats are not affected.

Filament load still calls setTargetHotend which the last commit to menu_temperature.cpp stopped using. I believe this is where the problem started.

### Bug Timeline

I believe this may have been introduced 3 months ago.

### Expected behavior

Selected hotend for preheat should be the one targeted.

### Actual behavior

Only active extruder can be preheated when using temperature preheat menu.

### Steps to Reproduce

Power on dual extrusion printer with current bugfix.
Preheat ABS or PLA Menu
Select Preheat END 2
First Hotend will heat.
Select other filament type
Select Preheat END 1
First Hotend temp will change.
Send T1
Observe opposite problem with above.

### Version of Marlin Firmware

Bugfix-2.1.x

### Printer model

Test on Replicator & BigBox

### Electronics

Stock MightyBoard or Rumba tested.

### LCD/Controller

FF_Interface or RepRap Full Graphic 

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[ReplicatorConfig.zip](https://github.com/user-attachments/files/20813505/ReplicatorConfig.zip)

ellensp, 2025-06-19:
Tested in the simulator, also heats wrong extrtuder

ellensp, 2025-06-19:
The menu is

[image]

As I read the code... 
Preheat PLA 1           should heat up both hotends to PLA preset. it only heats up hotend 1
Preheat PLA END 1  should heat up hotend 1 and it does
Preheat PLA 2          should heat up both hotend to PLA preset. it only heats up hotend 2, and why are there two of these?
Preheat PLA END 2 should heat up hotend 2 and it heats up 1

This is very confusing.

RainMotorsports, 2025-06-19:
> This is very confusing.

I don't entirely understand the code but in this version https://github.com/MarlinFirmware/Marlin/blame/fc49efcaef97ab657b28ce1077c7bc07d538bacc/Marlin/src/lcd/menu/menu_temperature.cpp

Before the current commit the preheat for hotend is:
if (targets & PreheatTarget::HOTEND) setTargetHotend(ui.material_preset[m].hotend_temp, e);

This uses a similar setup to what I saw in the menu_filament.cpp regarding preheating a particular hotend which we know works. I haven't had a chance to test this older commit or just before the 3 commits that affected this part of the code.

The current commit has the code like this:
inline void _preheat_end(const uint8_t m, const uint8_t e) { ui.preheat_hotend(m, e); ui.return_to_status(); }
    void do_preheat_end_m() { _preheat_end(editable.int8, 0); }

Which is quite different and has me searching this area but I have stopped testing for now.

ellensp, 2025-06-19:
I went back to a older version and yes
```
Preheat PLA/ABS 1              heats the hotend 1 and the bed,   works as expected
Preheat PLA/ABS END 1     heats the hotend 1                       ,  work as expected 
Preheat PLA/ABS 2              heats the hotend 2 and the bed ,  work as expected 
Preheat PLA/ABS END 2     heats the hotend 2                       ,  heats up hotend 1

```

ellensp, 2025-06-20:
I will do it.