MarlinFirmware/Marlin #27935: [BUG] Smooth Linear Advance results in underextrusion curve

vovodroid, 2025-06-21:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

**Smooth Linear Advance with extruder synchronisation #27710** introduces underextusion, depending on extrusion speed. Extrusion length is measure with long (500mm) cold extrusion without nozzle.

Configuration used:

```
#define LIN_ADVANCE
#define ADVANCE_K 0.22        // (mm) Compression length applying to all extruders
#define SMOOTH_LIN_ADVANCE    // Remove limits on acceleration by gradual increase of nozzle pressure
#define ADVANCE_TAU 0.02       // (s) Smoothing time to reduce extruder acceleration
#define SMOOTH_LIN_ADV_HZ 1000   // (Hz) How often to update extruder speed
#define INPUT_SHAPING_E_SYNC     // Synchronize the extruder-shaped XY axes (to increase precision)
//#define NONLINEAR_EXTRUSION
```

Here is underextrusion table for speed (mm/s):
```
3	0.929
6	0.926
18	0.985
30	0.968
39	0.944
```

[image]

No LA/Tau change affect this. Fast 500MHz SKR-3 board with TMC2209 is used.

### Version of Marlin Firmware

butfix-2.1.x

### Electronics

SKR-3, TMC2209

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/user-attachments/files/20847500/Marlin.zip)

dbuezas, 2025-06-22:
It could also be e-synch. Can you confirm none of your input shaping frequencies are below the smallest one set at build time (i.e the default IS frequencies)?

vovodroid, 2025-06-22:
Or do you mean that they are less than 58/28?

dbuezas, 2025-06-22:
I assume with the same results?
Did you try disabling e_synch?

dbuezas, 2025-07-09:
I finally got around testing it myself, the issue is not affecting my machine.
I used the same speeds as you. There's only underextrussion at 39mm/s, which is anyway an extreme speed.

| mm/s | mm/min | Flow (mm³/s) | mm1    | mm2    |
|------|--------|--------------|--------|--------|
| 39   | 2340   | 93.80        | 99.50  | 99.85  |
| 30   | 1800   | 72.15        | 101.73 | 101.43 |
| 18   | 1080   | 43.29        | 101.60 | 101.10 |
| 6    | 360    | 14.43        | 101.15 | 101.70 |
| 3    | 180    | 7.22         | 101.74 |  101.50 |

vovodroid, 2025-07-10:
Any difference in our configuration could lead to such results?

dbuezas, 2025-07-10:
I see you have `S_CURVE_FACTOR` enabled so I suppose you are using #27101. That's the only thing I see that could affect it.
SmoothLA uses specific code to integrate with "normal" S_CURVE, so maybe something went wrong there.
I suggest you to make a test from latest bugfix branch to be sure

vovodroid, 2025-07-14:
Ok, I retested without S_CURVE (any type), INPUT_SHAPING and NONLINEAR_EXTRUSION. The same result - 3-4% underextrusion an 3mm/s speed.

dbuezas, 2025-07-14:
I assume you u are testing directly from bugfix, right?
Also with NLE disabled during build.
 I don't know what could be the issue, I can't reproduce it :/

dbuezas, 2025-07-14:
I use an skr3, I do have SCurve enabled (but not the special one in an open MR)
The only thing I can think of is trying higher micro stepping to check if some fractional error could yve related to it

dbuezas, 2025-07-14:
It is just a guess, if the problem goes away at higher microsteps it would be a good clue

vovodroid, 2025-07-15:
I tried with 32 microsteps and 806 steps/mm. The same.