MarlinFirmware/Marlin #27937: [BUG] LCD_BED_TRAMMING does not respect Z_MAX_POS

RainMotorsports, 2025-06-22:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

LCD_BED_TRAMMINGs first move after homing is current_position.z + BED_TRAMMING_Z_HOP which on printers that home to Z Max effectively becomes Z_MAX_POS + BED_TRAMMING_Z_HOP. In my case the endstop is 200 and it tries to make a move to 204. After skipping it steps it then does the blocking XY move to the first position and then because I am not using a probe it moves to BED_TRAMMING_HEIGHT as it should. Subsequent moves are fine as current z position is BED_TRAMMING_HEIGHT in this case 0.

The first workaround I could think of is `Z_AFTER_HOMING` but this does not work with Z Max Homing. The only other workaround so far is `ENDSTOPS_ALWAYS_ON_DEFAULT` as suggested by @thisiskeithb .   This does work but isn't necessarily ideal.

Looking at the code it seems current_position.z may be necessary for use with probes. Otherwise my simplest and most elegant solution was to replace `line_to_z(z);` with `line_to_z(BED_TRAMMING_Z_HOP);` possibly throwing BED_TRAMMING_HEIGHT into the mix if necessary.

My first thought is to check that `current_position.z + (BED_TRAMMING_Z_HOP)` does not exceed Z_MAX_POS before assigning it as the code does in `float z = current_position.z + (BED_TRAMMING_Z_HOP);`

### Bug Timeline

Likely 2+ years if not longer.

### Expected behavior

LCD Bed Tramming should respect Z_MAX_POS

### Actual behavior

LCD Bed Trammings first move after homing on a printer homed to z max is to exceed z max by the tramming z hop value.

### Version of Marlin Firmware

Bugfix-2.1.x

### Printer model

MakerGear M2 Rev H

### Electronics

Stock Rambo 1.3L

### LCD/Controller

REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

### Bed Leveling

No Bed Leveling

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Config.zip](https://github.com/user-attachments/files/20851717/Config.zip)

RainMotorsports, 2025-06-23:
Replacing
```
float z = current_position.z + (BED_TRAMMING_Z_HOP);
```
With
```
float z =  _MIN(current_position.z + (BED_TRAMMING_Z_HOP), Z_MAX_POS);
```

Resolves the issue for me. I found this method in the M701/702 code for doing a park raise with limit to Z max. This should be probe safe as they would never run into a situation where current z position + hop is greater than the z max position.

thisiskeithb, 2025-06-23:
> Resolves the issue for me

You should open a PR for an official review 🙂

closed 2025-07-19 by thinkyhead