MarlinFirmware/Marlin #27942: [BUG] Power Loss Recovery incorrectly labeled in Creality DWIN menu.

TheAndr0id, 2025-06-27:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

This started as a desire to have a preheat setting for PETG and to remove the "Preheat ABS".  After many hours of digging it seems that the "PLA" and "ABS" components are actually icons, not text and the PREHEAT_*_LABEL config items are not used.

After more digging, it seems they ***are*** used, if USE_STRING_TITLES is defined.  Ok....

But, there's a problem/bug.  Take a look at src/lcd/e3v2/creality/dwin.cpp around line 2500:

```
void itemAdv_PLR(const uint8_t row) {
    if (false && hmiIsChinese()) {
      // TODO: Chinese "Power-loss Recovery"
    }
    else {
      #ifdef USE_STRING_TITLES
        dwinDrawLabel(row, GET_TEXT_F(MSG_ZPROBE_OFFSETS));
      #else
        itemAreaCopy(1, 208, 137, 221, row);  // "Power-loss Recovery"
      #endif
    }
```

If you don't see it, MSG_ZPROBE_OFFSETS is not "Power-loss Recovery".  Someone copy'n'pasted....

I added a MSG_POWERLOSS_RECOVERY to language_en.h, and then changed the value above.

This change needs to be placed into the code.

Also, why is USE_STRING_TITLES not an item in the configuration.h?  

### Bug Timeline

Unknown.  

### Expected behavior

To have the right label for the menu...

### Actual behavior

"Probe Offsets" != "Power Loss Recovery"

### Steps to Reproduce

1. Look at code.
2. Realize copy'n'paste error

### Version of Marlin Firmware

Found in 2.1.2.5, but is also in the current Bugfix and github.

### Printer model

Ender 3V2

### Electronics

Stock

### LCD/Controller

DWIN (Stock)

### Other add-ons

None.

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2025-06-27:
This will be fixed in https://github.com/MarlinFirmware/Marlin/pull/27943

thisiskeithb, 2025-06-27:
> Also, why is `USE_STRING_TITLES` not an item in the `Configuration.h`?

Not all options are exposed in the config files, but I've opened a PR to add them:

- #27944

closed 2025-07-19 by thinkyhead