MarlinFirmware/Marlin #27948: [FR] Support for SDCARD to microSD card adapters (Ender 3* + more)

TheAndr0id, 2025-06-28:
### Is your feature request related to a problem? Please describe.

Many printers(Notably Ender 3s) use a microSD card port on their motherboard.  microSDs are fine for adding storage that is *almost never* removed, but for a printer not so much.  There are also concerns with the manual handling of microSD cards, the load/eject mechanism wear, and having to use adapters inside of adapters to access them on a PC.  The solution is to use an adapter such as this (Image stolen from Amazon):

[image]

The issue is Marlin doesn't see any card changes after the system boots.  Card rereads are triggered off a `SD_DETECT_PIN` and these adapters don't have a `SD_DETECT_PIN`.  So the only way to get the system to read the SD card is to cycle power or to remove and insert the adapter - which is silly.  This is not good for the power supply and the printer's components.

### Are you looking for hardware support?

Yes and no.  There is no new hardware to support, just how it's utilized.  This *should* be doable within the context of the current code base with no additional hardware support other than defining the "adapter" as hardware with a possible define flag in `configuration.h`.

### Describe the feature you want

When the Print Menu/File Browser is selected the contents of the current SD card should be read and displayed.  If the card is removed, no files should be shown (no media present message), and when a new SD card is inserted the file list should be updated when the file browser is updated.

Changes to the file browser don't need to be updated dynamically.  If a new card is inserted, selecting "back" and then opening the browser is very acceptable.

This change should work no matter what UI is being used (DWIN/Marlin/etc.)

### Additional context

Some of this has been attempted in the past to limited success.  Here's what seems to be going on, and if incorrect, I apologize: 

Most boards use a `SD_DETECT_PIN` (eg: Ender 3s).  When this pin's state changes `IS_SD_INSERTED` reads the pin's state and compares against `SD_DETECT_STATE` (which can also be changed in `configuration_adv.h`).

`NO_SD_DETECT` can be found in `configuration_adv.h`, but all that does is undefines `SD_DETECT_PIN` which then hard defines `IS_SD_INSERTED` to true.

The first time `card.manage_media()` is called(at boot), `IS_SD_INSERTED` causes the card to be mounted and read.  This change of status is kept in the static variable `stat`, and whenever `card.manage_media()` is called again, the `stat` variable is not showing any changes and thus there's no need to reread or update the file listing.

Unless `IS_SD_INSERTED` changes (which is triggered by the `SD_DETECT_PIN` or in the case of `NO_SD_DETECT` never changes) `card.manage_media()` will always short exit out of the function.

Some options that may work:

1. A reddit post [Enable simple swapping of SD cards when using an extender](https://www.reddit.com/r/ender3v2/comments/lx3vko/enable_simple_swapping_of_sd_cards_when_using_an/) by _Kooky-Ad1883_ describes adding a force flag to manage_media() to ignore the previous state of the `stat` variable.  Then adding a call to `card.manage_media(true);` into the routine that draws the browser to force the card's contents to update.
2. Setting `IS_SD_INSERTED` to a timed flip flop from false to true.  This would cause the card to be polled on a regular basis.  This would be a challenge to balance load vs timing vs SD card wear.
3. Add a menu item to the top of the browser forcing a SD card update.  Basically as if an [M20](https://marlinfw.org/docs/gcode/M020.html) were issued.
4. ???

TheAndr0id, 2025-06-28:
This is a copy of the above mentioned reddit article by Kooky-Ad1883 
(Just in case it goes away)

https://www.reddit.com/r/ender3v2/comments/lx3vko/enable_simple_swapping_of_sd_cards_when_using_an/

> 
> ### Enable simple swapping of SD cards when using an extender
> 
> I use an SD card extender and have a few SD cards. It irritated me that I had to remove and re-insert the SD card extender in order to get the new SD card to be recognized so I decided to have a dig into the firmware. The simplest solution was to get the SD card to be mounted each time you access the print menu so that you can swap SD cards and then access the print menu to see the new contents. I posted the code changes elsewhere but I have been asked to create a dedicated post so here it is in case it is of use to others.
> 
> There are 3 files that need changing, Marlin/src/lcd/dwin/e3v2/dwin.cpp, Marlin/src/sd/cardreader.cpp and Marlin/src/sd/cardreader.h
> 
> dwin.cpp - Find the function Draw_Print_File_Menu() and add the line shown in bold below:
> inline void Draw_Print_File_Menu() {
> Clear_Title_Bar();
> **card.manage_media(true);**
> 
> cardreader.h - Find the declaration static void manage_media(); and make the changes in bold:
> static void manage_media(**const bool force=false**);
> 
> cardreader.cpp - Find the function manage_media() and make the changes in bold:
> void CardReader::manage_media(**const bool force**) {
> static uint8_t prev_stat = 2;       // First call, no prior state
> uint8_t stat = uint8_t(IS_SD_INSERTED());
> if (**!force &&** (stat == prev_stat)) return;
>

ellensp, 2025-06-28:
Marlin is a g-code controller

you use gcodes M21/M22

TheAndr0id, 2025-06-28:
Just when I thought I was out, it pulled me back in.

The M20 got me thinking, so I went and found the M2[012].cpp files and did a read.

The basics (there are condition checking abounds) are:
```
M20:   calls card.ls();
M21:   calls card.mount();
M22:   calls card.release();
```
And just doing some testing over the serial USB, calling M21 after ejecting the SD before opening the browser seems to do the trick.  I'm going to try adding a
`card.mount();` into `Draw_Print_File_Menu()`

### **And that works.  That was easy.**

Reviewing card.mount() it also does a `if (root.isOpen()) root.close();` so there shouldn't need to be a `card.release()` call either. (I think!)

I'm not pretending to totally understand the consequences of using `card.mount()` over and over.  Comments?

It still needs some logic wrapped around the call and to have it work with the other UIs.

TheAndr0id, 2025-09-16:
I added the card.mount into the drawPrintFileMenu() function instead of when the selector is triggered.  I should have taken a look at the ProUI code as many of these issues have been assessed there.  One of these days I'll give up and switch to either Marlin or ProUi, I'm just being stubborn and sticking with the factory look...

I think I'm going to add the 800ms(?) delay in as it's probably a good idea.

```
void Draw_Print_File_Menu() {
  Clear_Title_Bar();

  card.mount();

  if (HMI_IsChinese())
    DWIN_Frame_TitleCopy(0, 31, 56, 14);  // "Print file"
  else {
```

I've been running with this for a few months now with no issues with file/SD card access.

Does the MarlinUI handle these extenders as the ProUI does?

TheAndr0id, 2025-09-20:
I just tested adding a 800ms delay to the mount.  Maybe I'm just used of it not being there, but I don't like it and I'm going to remove it.  

The mount() command should be able to handle waiting for the card to be ready.  Only the ProUI interface introduces the 800ms delay after the mount() command.  While I didn't do an exhaustive search, the first 5-6 instances of mount() commands I did check had no delay after their call.

I don't think it's anything that's dangerous just unnecessary.