MarlinFirmware/Marlin #27955: [BUG] M109/M190 commands do not clear status message

TheAndr0id, 2025-06-29:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The two commands M109 (hotend) and M190(bed) generally are used to wait until the hotend/bed get up to temperature.  They both update the status message stating they are heating (they use different methods, M109 uses thermalManager.set_heating_message() and M190 seems to use a LCD_MESSAGE() macro...).

As expected both commands wait until the temp is reached and exit, but the message that the hotend/bed is still heating remains on the status display.

### Bug Timeline

No changes to M109 or M190 for over 2 years.  Old bug?

### Expected behavior

After M109/M190 exits, the status message should be cleared.

### Actual behavior

M109/M190 commands leave a remnant of their status on the display that is incorrect.  

### Steps to Reproduce

Use M109/M190 with the status display enabled.

### Version of Marlin Firmware

2.1.2.5, but the code is the same in the 2.1.3-beta3

### Printer model

Ender 3v2

### Electronics

Stock

### LCD/Controller

stock DWIN

### Other add-ons

none

### Bed Leveling

None

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I think this can be easily worked around with a M117 with an empty message after the M109/M190 calls in startup gcode.    Adding empty M117 calls after every M109/M190 (think print pauses) is an ugly pain to fix something obvious.

I haven't included configuration files as these commands are basic functionality that almost all configurations will include....

thisiskeithb, 2025-06-29:
This behavior is intentional. If you want the status screen to reset after a desired amount of time, then enable & adjust `STATUS_MESSAGE_TIMEOUT_SEC`.

closed 2025-06-29 by thisiskeithb

TheAndr0id, 2025-06-29:
I would agree with the use of a standard M117 call, or even with a M104/M140 where the actions are basically fire and forget, but M109 and M190 are active (and blocking) monitoring of the temperature.  

This is kind of critical and the status display should reflect that the monitoring is happening and when it is complete.  If the display states that the hotend is heating, I expect that to be true, not a "well maybe it is, maybe it isn't."

We are talking about stuff that can burn your house down if it goes sideways.  Myself I'd want to make sure at least the status display was accurate.  Just sayin'...

TheAndr0id, 2025-07-16:
After looking at the code, this is in fact a bug that needs to be addressed.  In 
[Marlin/src/module/temperature.cpp](https://github.com/MarlinFirmware/Marlin/blob/66474f6a1320c8e669a98e699eb819e1c8dd012b/Marlin/src/module/temperature.cpp#L4866) near line 4866:
```
 // If wait_for_heatup is set, temperature was reached, no cancel
      if (wait_for_heatup) {
        wait_for_heatup = false;
        #if ENABLED(DWIN_CREALITY_LCD)
          hmiFlag.heat_flag = 0;
          duration_t elapsed = print_job_timer.duration();  // Print timer
          dwin_heat_time = elapsed.value;
        #elif ENABLED(SOVOL_SV06_RTS)
          update_time_value = RTS_UPDATE_VALUE;
          if (card.isStillPrinting()) rts.refreshTime();
          rts.start_print_flag = false;
        #else
          ui.reset_status();
        #endif
```

This checks if the interface is DWIN or SOVOL and updates some variables for their environments.  Everyone **ELSE** gets a reset_status() call.  

This clearly should be for everyone.  Remove the #else and replace it with the #endif.

This might have been correct back in v1.02 or so when the DWIN environment didn't have a status message, but it does now.  (I can't speak for the SOVOL display environment...)