MarlinFirmware/Marlin #27972: Compile error on CR10V3 due to removed code

nicwilson58, 2025-07-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

in recent build some code has been removed in "Conditionals-2-LCD.g" under the section "// Flow and feedrate editing" starting on line 666.

The removal of this code causes a compile error when compiling for a Creality CR10V2 or V3:
"Marlin\src\lcd\marlinui.cpp:723:22: error: 'SPEED_EDIT_MIN' was not declared in this scope"

Replacing the removed code fixes the compile error.

Below is the code that was removed recently:

// Flow and feedrate editing
#if HAS_EXTRUDERS && ANY(HAS_MARLINUI_MENU, DWIN_CREALITY_LCD, DWIN_LCD_PROUI, MALYAN_LCD, TOUCH_SCREEN)
  #define HAS_FLOW_EDIT 1
#endif
#if ANY(HAS_MARLINUI_MENU, ULTIPANEL_FEEDMULTIPLY, DWIN_CREALITY_LCD, DWIN_LCD_PROUI, MALYAN_LCD, TOUCH_SCREEN)
  #define HAS_FEEDRATE_EDIT 1
#endif

### Bug Timeline

very recently,  dont know exactly when,

### Expected behavior

no compile error,   as stated its fixed by putting the removed code back.

### Actual behavior

compile error as stated.

### Steps to Reproduce

Compile for CR10V3 on latest code.

### Version of Marlin Firmware

very latest nightly build from 10-7-25

### Printer model

CR10V3

### Electronics

STOCK

### LCD/Controller

STOCK

### Other add-ons

N/A

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

No configs are attached as they are not needed,  this is a bug that affects all versions at least for similar printers.  Just check the removed code noted above.

thisiskeithb, 2025-07-10:
> No configs are attached as they are not needed, this is a bug that affects all versions at least for similar printers. Just check the removed code noted above.

Always follow the report template, including attaching the requested files.

Using the latest `bugfix-2.1.x` (dc6f23e) with stock and BigTreeTech SKR Mini E3 V3-based [Creality CR-10 V3 configs we host over in the Configurations repo](https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Creality/CR-10%20V3/Stock), there are no issues compiling for either config.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.com/servers/marlin-firmware-461605380783472640) Discord server

closed 2025-07-10 by thisiskeithb

nicwilson58, 2025-07-11:
so it worked on one build and that is as far as you go.   and so ends my support of marlin after 6 years.  From now on I report nothing,  I will just fix the code myself and not share it.

ellensp, 2025-07-11:
Your config is out of date, it is not a code issue

section

```cpp
#if HAS_FEEDRATE_EDIT
  #define SPEED_EDIT_MIN    10  // (%) Feedrate percentage edit range minimum
  #define SPEED_EDIT_MAX   999  // (%) Feedrate percentage edit range maximum
#endif
#if HAS_FLOW_EDIT
  #define FLOW_EDIT_MIN     10  // (%) Flow percentage edit range minimum
  #define FLOW_EDIT_MAX    999  // (%) Flow percentage edit range maximum
#endif
```

was replaced with

```cpp
#if ANY(HAS_MARLINUI_MENU, DWIN_CREALITY_LCD, DWIN_LCD_PROUI, MALYAN_LCD, TOUCH_SCREEN, ULTIPANEL_FEEDMULTIPLY)
  #define SPEED_EDIT_MIN    10  // (%) Feedrate percentage edit range minimum
  #define SPEED_EDIT_MAX   999  // (%) Feedrate percentage edit range maximum
#endif
#if ANY(HAS_MARLINUI_MENU, DWIN_CREALITY_LCD, DWIN_LCD_PROUI, MALYAN_LCD, TOUCH_SCREEN)
  #define FLOW_EDIT_MIN     10  // (%) Flow percentage edit range minimum
  #define FLOW_EDIT_MAX    999  // (%) Flow percentage edit range maximum
#endif
```

eliminating the need for HAS_FEEDRATE_EDIT and HAS_FLOW_EDIT

thisiskeithb, 2025-07-11:
> Your config is out of date, it is not a code issue

Bingo!

This is exactly why we ask users to test the latest `bugfix-2.1.x` **and** attach their configuration files when creating bug reports. This “bug” would have been easier to identify and remove all guess work. We can only test with what was provided (nothing in this case), so our hosted configs were used.

> so it worked on one build

Two.

> and that is as far as you go. and so ends my support of marlin after 6 years. From now on I report nothing, I will just fix the code myself and not share it.

Now, now. There’s no need to start violating the GPL because you’re running out of date configs on a newer version of Marlin and thought it was a bug.

Please use Facebook or Discord next time so we can determine if it’s a bug and save everyone the trouble.