MarlinFirmware/Marlin #27981: [BUG] Print Complete Status for DWIN issues (multiple)

TheAndr0id, 2025-07-16:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When a print completes the DWIN "print complete" popup has multiple issues.  They include:

1. Progress Bar shows 00% instead of 100% when complete.
2. Still updates Print Time Elapsed and Remaining when print is complete.
3. Print complete "box/window" is sized incorrectly.
4. The HMI_flag.print_finish (hmiflag.print_finish in v2.1.3-beta3) is not correctly set.  This seems to cause the print complete message to pop up while the printer is has a few commands left to do.

In general it kind of feels like DWIN is the interface time forgot.  It needs some general TLC.

### Bug Timeline

Old.  Seems to be an issue since v1.x

### Expected behavior

Cleaner display with the correct information.

### Actual behavior

It's just generally not correct.  Broken window sizes, 00% messages, print says it's finished and it's still doing stuff...

### Steps to Reproduce

Print something with DWIN interface.  Wait for it to complete.

### Version of Marlin Firmware

2.1.2.5, but these items are still in the current 2.1.3-beta3 code.

### Printer model

Creality Ender 3v2

### Electronics

Stock

### LCD/Controller

Stock DWIN

### Other add-ons

none

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Some of these look to be just simple edits to fix.  Some may have consequences I'm not aware of and need review.  As a reference to the numbered items above:

1. The display value is just set to 2 digits, but 100, is, well 3.  More I think of it, might be clearer with an if/else and a static string display in the case of 100%.
```void Draw_Print_ProgressBar() {
  constexpr uint16_t y = 93, h = 21;
  DWIN_ICON_Show(ICON, ICON_Bar, 15, 93);
  DWIN_Draw_Rectangle(1, BarFill_Color, 16 + _card_percent * 240 / 100, y, 256, y + h - 1);
// This doesn't work for 100%.
//  DWIN_Draw_IntValue(true, true, 0, font8x16, Percent_Color, Color_Bg_Black, 2, 117, y + 40, _card_percent);
  DWIN_Draw_IntValue(true, true, 0, font8x16, Percent_Color, Color_Bg_Black,  (_card_percent < 100) ? 2 : 3, (_card_percent < 100) ? 117 : 109, y + 40, _card_percent);
  DWIN_Draw_String(false, font8x16, Percent_Color, Color_Bg_Black, 133, y + 40, F("%"));
}
```
2. Since it seems the Goto_PrintProcess() function is still called even if the print is complete, the updates shouldn't happen if the print is complete.
```
if (!HMI_flag.print_finish){
    Draw_Print_ProgressElapsed();
    Draw_Print_ProgressRemain();
}
```
3. Seems the background box wasn't updated when the display size was increased.  It doesn't cover the print control buttons (that are no longer active).  The Y start location should be higher up (230 seems to be a good value).  In the `EachMomentUpdate()` function in `dwin.cpp`:  (Change the 250 to 230)
```
      // show print done confirm
      DWIN_Draw_Rectangle(1, Color_Bg_Black, 0, 250, DWIN_WIDTH - 1, STATUS_Y);
```
4. Looking at previous releases (some ancient including the last source Creality released - v1.02) the `HMI_flag.print_finish` (note `HMI_flag` has been renamed to `hmiflag`) boolean is just what it says - true if the print is finished.  That's not so consistent in the code today.

In `dwin.cpp` this causes problems in other functions that check `HMI_flag.print_finish`:
(I don't think it should be reset to false until the beginning of a new print/clearing of active job status)
```
if (checkkey == PrintProcess) {
    // if print done
    if (HMI_flag.print_finish && !HMI_flag.done_confirm_flag) {
      HMI_flag.print_finish = false;
      HMI_flag.done_confirm_flag = true;
```
In `cardreader.cpp`, the only place it is set is in the `endFilePrintNow` function, and it's value is less than clear depending on what `flag.sdprinting` could be (it's clear it should be set to true).  Also `flag.sdprintdone` isn't set here, why is `HMI_flag.print_finish`?
```
//
// Run tasks upon finishing or aborting a file print.
//
void CardReader::endFilePrintNow(TERN_(SD_RESORT, const bool re_sort/*=false*/)) {
  TERN_(ADVANCED_PAUSE_FEATURE, did_pause_print = 0);
  TERN_(DWIN_CREALITY_LCD, HMI_flag.print_finish = flag.sdprinting);
  flag.abort_sd_printing = false;
  if (isFileOpen()) file.close();
  TERN_(SD_RESORT, if (re_sort) presort());
}
```
The other interfaces use `flag.sdprintdone` to indicate the printing is complete, but `HMI_flag.print_finish` is never set when it is.  Perhaps doing this every place `flag.sdprintdone` is set?

`TERN_(DWIN_CREALITY_LCD, HMI_flag.print_finish = flag.sdprintdone);`

Finally M1001 is sent to indicate the print job is complete.  In the code:
```
/**
 * M1001: Execute actions for SD print completion
 */
void GcodeSuite::M1001() {
  planner.synchronize();

  // SD Printing is finished when the queue reaches M1001
  card.flag.sdprinting = card.flag.sdprintdone = false;
```

Once again, should `card.flag.sdprintdone` be set to false here?  The job is complete, and could setting it to false cause unexpected results in other functions?  If M1001 is the final command of a job, why clear it and take a chance?  (`HMI_flag.print_finish` isn't set either)

TheAndr0id, 2025-07-18:
After doing some testing (and I won't proclaim to know why exactly), but the setting of `HMI_flag.print_finish` and `card.flag.sdprintdone` to false in M1001 and when the print is complete in `dwin.cpp` are needed.

If they are set to true, the UI falls into some sort of confused race condition and (my guess only) that multiple M1001s are issued to the print planner which results in multiple "finished print" popups.  

I've updated the issue description for clarity.

TheAndr0id, 2025-07-20:
I _think_ I've figured out what is going on with the print complete box and a fix for it.

In the code from `dwin.cpp`:
```
if (checkkey == PrintProcess) {
    // if print done
    if (HMI_flag.print_finish && !HMI_flag.done_confirm_flag) {
      HMI_flag.print_finish = false;
      HMI_flag.done_confirm_flag = true;
```

The `if (HMI_flag.print_finish && !HMI_flag.done_confirm_flag)` condition is triggering if the print_finish flag is set, but there's no check to see if the printer is still printing.  Changing this line to:

`if (HMI_flag.print_finish && !card.isPrinting() && !HMI_flag.done_confirm_flag) {`

Stops the popup from coming up until after M1001 has completed.  This also stops the `Draw_Print_ProgressElapsed()`  and  `Draw_Print_ProgressRemain()` from updating the display after the print completed window pops up.

`HMI_flag.print_finish` is currently only set under `CardReader::endFilePrintNow()`:

`TERN_(DWIN_CREALITY_LCD, HMI_flag.print_finish = flag.sdprinting);`

This works since the function is called before the printer could process the queue, but there could(?) be a case where this could fail due to a race condition?  
(Aka, call `EndFilePrintNow()` and flag.sdprinting is false)

Once again, I've explicitly set `HMI_flag.print_finish` to echo the value of `card.flag.sdprintdone` throughout `cardreader.cpp`.

I hope this finally puts this annoying issue to rest.

TheAndr0id, 2025-08-02:
The TL;DR: Sorry, I won't get involved in development.

Here's the long story.

I totally support open source, almost religiously, but I've had some bad experiences in the past getting involved with development.  It was probably me, but the result was frustration and anger.

I swore over 20 years ago never to get involved again.

Now I believe firmly in giving something back.  When I find issues I try my best to provide not only a detailed description but also how *I* solved the problem.   My solution is not always the only way and I totally get that.  I also don't have the interest in starting another argument on the internet....

I *almost* did open the pull request.  Then I started to look at the requirements of the process and reminded myself why I stay out of the development projects.  I understand why you have the process in place, but frankly I want to create with my printer, not spend more time debugging it.

To simplify the initial issue report:

1) Instead of using the complicated conditional, just handle the case where _card_percent is 100.  In `Draw_Print_ProgressBar()` I did the following:

```
if (_card_percent < 100)
  DWIN_Draw_IntValue(true, true, 0, font8x16, Percent_Color, Color_Bg_Black, 2, 117, y + 40, _card_percent);
else
  DWIN_Draw_String(false, font8x16, Percent_Color, Color_Bg_Black, 109, y + 40, F("100"));
```
The x start (109) in the `DWIN_Draw_String()` will need to be checked to make sure it's spaced correctly.  The font is 8x16, thus the three characters before the % need 24 px to fit so 133-24=109 should work...  (+3 lines change)

2) This isn't needed.  The fix in item 4 below negates this.

3) Change 250 to 230 in the `DWIN_Draw_Rectangle` call in `EachMomentUpdate()` as described. (+1 line change)

4) My solution in the comment above seems to have fixed the issue with the print complete showing before the job is finished.  The main change is checking the printing status `card.isPrinting()` in the `EachMomentUpdate()`:
```
if (checkkey == PrintProcess) {
    // if print done
    if (HMI_flag.print_finish && !card.isPrinting() && !HMI_flag.done_confirm_flag) {
      HMI_flag.print_finish = false;
      HMI_flag.done_confirm_flag = true;
```
(+1 line change)

There are also numerous places in `cardreader.cpp` where `card.flag.sdprintdone` is set and `HMI_flag.print_finish` should also be set.    Add the following line after each instance:

``
TERN_(DWIN_CREALITY_LCD, HMI_flag.print_finish = card.flag.sdprintdone);
``
(multi-line search/copynpaste)

`HMI_flag.print_finish`  is also set in `CardReader::endFilePrintNow()` and I think it should be removed.  It works(?) because of a race condition, but without using a full debug I can't confirm that.  The logic is complicated at best and dodgy at worst.  (+1 change)

Total 6 lines of change + a 2 minute search/paste on cardreader.cpp. 
( I already have the diff/patches to apply to 2.1.3 when it's released...  😄 )

classicrocker883, 2025-08-20:
have you tried using ProUI?
i have [my own fork here](https://github.com/classicrocker883/MRiscoCProUI)

if youd like to check out some [precompiled files](https://classicrocker883.github.io/firmware-selector) you can select from

I believe I have these fixes

classicrocker883, 2025-08-20:
try this for the progress bar - in **Marlin/src/lcd/e3v2/creality/dwin.cpp**

add `#include "../../../libs/numtostr.h"` to the top of file

```
void drawPrintProgressBar() {
  constexpr uint16_t y = 93, h = 21;
  dwinIconShow(ICON, ICON_Bar, 15, 93);
  // dwinDrawRectangle(1, COLOR_BARFILL, 16 + _card_percent * 240 / 100, y, 256, y + h - 1);
  // dwinDrawIntValue(true, true, 0, font8x16, COLOR_PERCENT, COLOR_BG_BLACK, 2, 117, y + 40, _card_percent);
  // dwinDrawString(false, font8x16, COLOR_PERCENT, COLOR_BG_BLACK, 133, y + 40, F("%"));
  dwinDrawRectangle(1, COLOR_BARFILL, 15 + (_card_percent * 242) / 100, y, 257, y + h - 1);
  dwinDrawString(false, font8x16, COLOR_WHITE, COLOR_BG_BLACK, 117, 133, pcttostrpctrj(_card_percent));
}
```

TheAndr0id, 2025-09-27:
I'm sorry, I forgetting to come back and update this issue.  I modified(to work with 2.1.2.5) and tested your code, and it looks perfect.  Thanks!