MarlinFirmware/Marlin #27983: [BUG] Multiple Z endstops homming issue (with possible fix)

sorgelig, 2025-07-20:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Those who use printer with more than 2 Z endstops know about this long standing bug if VALIDATE_HOMING_ENDSTOPS enabled. I wanted to dig finally into this issue as it's quite annoying. It was especially interesting as 2 endstops are working bot not 3 or 4. 
In short: the issue happens around endstop adjusting values.
For 2 endstops there is a check if adj value is 0 and then last homing call get skipped. That's why 2 endstops are working. For 3 and 4 endstops homing get called even difference in endstop adjusting values are 0. I believe the fix should be more complex but for vast majority of users who don't use endstop adjusting the fix is simple. I've added diff file for v2.1.2.5 fix. Basically i've added test for 0 and skip redundant homing if true.
I guess the problem is in hit_status which get cleared on previous success homing. And then homing get called with 0 move, the hit_status not updated and remains 0 which triggers panic on home validation and eventual kill().
I believe home validation should be skipped if homing get called with endstop adjusting values as it's supposed to back-off from endstop and validation supposed not to pass anyway. Probably it shouldn't be homing but simple stepper move to specific amount. So real fix should include more flags and checks.

### Version of Marlin Firmware

2.1.2.5 (2.1.3-b3 as well)

### Printer model

DIY Core XY

### Electronics

BTT Octopus Pro v1.0

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

```
diff --git a/Marlin/src/module/motion.cpp b/Marlin/src/module/motion.cpp
index 0ff5db4..5130e8b 100644
--- a/Marlin/src/module/motion.cpp
+++ b/Marlin/src/module/motion.cpp
@@ -2176,25 +2176,25 @@ void prepare_line_to_destination() {
             if (pos_dir) {
               // normalize adj to smallest value and do the first move
               (*lock[0])(true);
-              do_homing_move(axis, adj[1] - adj[0]);
+              if(adj[1] - adj[0]) do_homing_move(axis, adj[1] - adj[0]);
               // lock the second stepper for the final correction
               (*lock[1])(true);
-              do_homing_move(axis, adj[2] - adj[1]);
+              if(adj[2] - adj[1]) do_homing_move(axis, adj[2] - adj[1]);
               #if NUM_Z_STEPPERS >= 4
                 // lock the third stepper for the final correction
                 (*lock[2])(true);
-                do_homing_move(axis, adj[3] - adj[2]);
+                if(adj[3] - adj[2]) do_homing_move(axis, adj[3] - adj[2]);
               #endif
             }
             else {
               #if NUM_Z_STEPPERS >= 4
                 (*lock[3])(true);
-                do_homing_move(axis, adj[2] - adj[3]);
+                if(adj[2] - adj[3]) do_homing_move(axis, adj[2] - adj[3]);
               #endif
               (*lock[2])(true);
-              do_homing_move(axis, adj[1] - adj[2]);
+              if(adj[1] - adj[2]) do_homing_move(axis, adj[1] - adj[2]);
               (*lock[1])(true);
-              do_homing_move(axis, adj[0] - adj[1]);
+              if(adj[0] - adj[1]) do_homing_move(axis, adj[0] - adj[1]);
             }
```

ellensp, 2025-07-20:
There is an open issue regarding this https://github.com/MarlinFirmware/Marlin/issues/23227 

, opening a second issue does not help things. please add this to the original issue.

closed 2025-07-20 by ellensp

thinkyhead, 2025-10-02:
Thanks for the report. I am checking it out now. Please be nice to the volunteers who triage these issues. They do a great job service to try to reduce the amount of work that falls into my lap. And it has been a challenging couple of years for this project.