MarlinFirmware/Marlin #27985: [BUG] CR6-SE X-Axis moves larger distance when spanning across the middle of the build plate

CJM3407, 2025-07-21:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

I did large spaces and noticed when I told my printer to move 170mm it was actually moving 175mm. I did the calculations to set my steps per mm from 80step/mm to 77.7steps/mm. This fixed the large distance of 170mm. But what I noticed was smaller distances were now too small.

So I set it back to the 80/mm and attached a sharpie to my head unit, and made my printer move 20mm increments in the x-axis. This revealed the center of my x-axis is doing a weird larger jump set, it goes fine for the first 70-80mm, then the gap is about 5% larger for the next 60-80mm, and the far right side is back the proper 20mm gaps. It seems to be consistent, so I don't think it is purely slips or mis-steps.

As for when it started, I was printed a larger game storage solution, that was 4 pieces. I printed the first tray of 4 pieces, and it all came out perfect. I started on the second tray, and the size of the first one was correct measurements at the base (expected113mm, got113mm). This print was interrupted by needing a filament changeover (during the night). In the morning, I did the switch, let it resume. I am now noticing the very tail end of print, is actually the 115mm in the x direction at the very top (second picture). So, this is probably around the time of the failure

The second quarter started up, finished, I put it next to Q1, and they are different width. Now i was measure 115mm compared to the supposed 113mm. I go to re-print, and my y-axis belt fails, I thought it snapped (about 8 months ago, I had to replace my x-axis belt because it breaking). It actually just came off, but from the replacement kit, I still had a y-axis belt, so I went ahead and replaced with the new one.

I print the rest of the pieces, and they all are too big, even going back and printing the original Q1 piece, and it is now big (115mm).

Which finally brings me to the spacial testing with a sharpie and to this post.

At this point, I have tried 6.1 a version 5x. As well as, I reset back to two different crealty firmwares (marlin 2.0.6 based), and the error is still happening.

### Expected behavior

When I tell my printer to move 170 mm it should move 170 mm

### Actual behavior

It moves about 175mm

### Steps to Reproduce

1. Home the printer
2. Tell the printer to move to 10mm x-axis. 
3. Tell the printer to move to 200mm x-axis
4. Measure the distance, and see it will be more like 205mm.

### Version of Marlin Firmware

CR6 Community version 6.1 as well as Crealty v2.0.1.9 (Marlin 2.0.6)

### Printer model

CR-6 SE

### Electronics

Stock

### LCD/Controller

Stock touch screen

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[image]
[image]

[configs.zip](https://github.com/user-attachments/files/21339429/configs.zip)

ellensp, 2025-07-21:
Provided Configs are for a positively ancient version of marlin 02000801 this is from May 16, 2021

ellensp, 2025-07-21:
Since its not code from official marlin repository you have to contact Creality support.

closed 2025-07-21 by ellensp