MarlinFirmware/Marlin #27986: [BUG] Input shaping problen on Y axis

MihranGalstyan, 2025-07-22:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I’m using a custom 3D printer with:

MKS SKR 1.4 Turbo

TMC2209 drivers

Marlin 2.1.2.5

Input Shaping works perfectly on the X-axis. However, when enabling it on the Y-axis, the bed moves very stiffly and jerks. During printing, steps are lost and objects shift.

Swapped drivers → no change.

Reassigned Y to E1 (second extruder port) → no change.

Swapped X/Y stepper plugs → issue moves to X.

Disabled StealthChop → no effect.

This suggests the issue is not hardware-related but possibly firmware or pin mapping behavior.

### Expected behavior

Input Shaping should work on Y-axis without causing step loss or jerky motion.

### Version of Marlin Firmware

2.1.2.5

### Printer model

Custom

### Electronics

SKR 1.4 TURBO

### LCD/Controller

RepRap Full Graphics

### Bed Leveling

None

### Your Slicer

Other (explain below)

### Host Software

Pronterface

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Is this a known issue with SKR 1.4 Turbo and TMC2209? Are there firmware settings or workarounds to resolve this?

patovers, 2025-09-24:
If this was fixed in a bug fix version as stated above, then it's been broken again since. I had this problem on a brand new installation of 2.1.2.5 and independently tracked it down to input shaping on the y axis. After installing the latest bug fix build this week I discovered that the shifting was still there but shifted by about half the distance (about 2mm instead of around 4-5). The shifting is exactly the same on each loop of the perimeter of the model, so it's clearly a systematic problem.

handyrandyrc, 2025-09-24:
Does it manifest something like this?
[image]

dbuezas, 2025-09-25:
https://github.com/MarlinFirmware/Marlin/blob/fdac56da1519e48459a89a4608a0451787b076d3/Marlin/Configuration_adv.h#L1240

SHAPING_MIN_FREQ

dbuezas, 2025-09-25:
If you don't set it lower than that later via menu or gcode, then yes, it's ok. I don't know what could cause that zigzag pattern, I've never seen that in the past

MihranGalstyan, 2025-10-02:
> 2.1.2.5

I had problem with stable 2.1.2.5 version. Installed **bugfix** version from 18.08.25, an there was no more input shaping problem with Y axis, but shaper was working not so correct.

tombrazier, 2025-10-06:
@MihranGalstyan Could you describe in more detail what you meant by "Swapped X/Y stepper plugs → issue moves to X."? This could be key information.

Also, anyone who has time to experiment, what happens if you disable X input shaping and just use Y?

patovers, 2025-10-06:
Here are my two configuration files. 
[Archive.zip](https://github.com/user-attachments/files/22728391/Archive.zip)

I can also confirm that the layer slipping is present when Y input shaping is enabled, irrespective of whether X input shaping is enabled or not.

michelmarlin, 2025-12-12:
Thank you Nuck-TH!! 
I will do but I need first to setup config and config_adv, ill report if it works tommorow!

michelmarlin, 2025-12-15:
After a few prints with IS enabled running bugfix, my good old Tevo Tornado with upgraded board and drivers printed the best looking parts of his ancient history, very smooth surface.
No missing steps anymore but still the Y axis sounds a bit more rough compared to the X axis.
If you swap X and Y and test the movement the physical X stepper is making more noise and the Y is more quiet.
Hopefully this can be improved in the future, if the X can be quiet why not the Y?
The printer is always in my sight for safety reasons, and the sound is a bit annoying.
But it finally prints fine, thats the best part.

patovers, 2026-02-03:
I finally got round to looking at this again. I have done a side by side comparison of @michelmarlin 's config files with my own and there is nothing really notable, though there are plenty of differences, e.g. BL touch vs magnetic sensor on my machine. The biggest differences I would say are:
Sensorless homing is used on mine.
Emergency parser enabled on mine.
Firmware retraction enabled on mine (though I don't think it was when I first hit this problem). 
Linear advance enabled on yours but not mine.
Input shaping settings are identical except for frequencies and damping.

I have also worked through various versions (2.1.2.5, 2.1.x, bugfix-2.1.x) all of which exhibit the Y slipping problem. I then decided to try 2.1.3 beta3, which is interesting because it doesn't show the slipping problem on the Y axis but it DOES on the X axis! In other words I can run with Y shaping enabled but X disabled without issue, whereas all the other version I can run with X shaping enabled but Y disabled. 
BTW I skipped 2.1.2.7 (the current latest release) because I can't even get it to compile. It throws these errors:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:3339:15: error: missing binary operator before token "("
 3339 | #if DGUS_UI_IS(MKS)
      |               ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:3341:17: error: missing binary operator before token "("
 3341 | #elif DGUS_UI_IS(IA_CREALITY)
      |                 ^

I will look into FT_MOTION next. I have already spent many hours making all these versions compile due to the pointless differences in define names between the versions. In 2.1.2.7, the CONFIGURATION_H_VERSION needs to be set to 02010206, and not ...07 so it makes me wonder how much testing is actually done on this stuff before a release!

ellensp, 2026-02-04:
re error

```
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:3339:15: error: missing binary operator before token "("
 3339 | #if DGUS_UI_IS(MKS)
      |               ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:3341:17: error: missing binary operator before token "("
 3341 | #elif DGUS_UI_IS(IA_CREALITY)
      |                 ^

```
@patovers this is the error you get when you use 2.1.3 configs (ie bugfix or the betas) on older code

The macro DGUS_UI_IS  only exists on  in 2.1.3

ie Coonfiguration.h has
```cpp
//#define DGUS_LCD_UI ORIGIN
#if DGUS_UI_IS(MKS)
  #define USE_MKS_GREEN_UI
#elif DGUS_UI_IS(IA_CREALITY)
  //#define LCD_SCREEN_ROTATE 90          // Portrait Mode or 800x480 displays
  //#define IA_CREALITY_BOOT_DELAY 1500   // (ms)
#endif
```

On older code it uses

```cpp
//#define DGUS_LCD_UI_ORIGIN
//#define DGUS_LCD_UI_FYSETC
//#define DGUS_LCD_UI_HIPRECY
//#define DGUS_LCD_UI_MKS
//#define DGUS_LCD_UI_RELOADED
#if ENABLED(DGUS_LCD_UI_MKS)
  #define USE_MKS_GREEN_UI
#endif
```

ppmanzionna, 2026-03-14:
I have the same issue on my printer with version 2.1.2.7 and earlier versions. By disabling input shaping for the Y‑axis, the problem does not occur.

Unfortunately, with the versions after 2.1.2.7, since INTEGRATED_BABYSTEPPING is no longer present (because it’s already integrated), babystepping no longer works for me, and I really can’t do without it.

[Configuration.zip](https://github.com/user-attachments/files/25993310/Configuration.zip)

[image]

patovers, 2026-03-27:
I'm happy to try out suggestions from developers but I don't have time for endless builds and test prints. I'm also happy to take patched versions of individual files and plug them into builds, provided that it's reasonably controlled and targeted. I'm not an experienced s/w developer but I AM an experienced h/w developer (ex Arm and many other companies). If I were in the position to try debugging this at the h/w level, then I'd be looking for what goes into the shaper and what comes out to see the differences. This may be a naïve view but my expectation is that without the shaper enabled (whichever one that is), then it should be a simple case of showing that output == input under all conditions to begin with, including looking at the HAL code, and then enabling it with a range of frequency and damping factors and seeing where output != input but is expected to match. The other thread, i.e. is it a difference in the HAL for the two boards known to differ in their behaviour, means we need to look at the step output controls for differences, especially if we are to assume that two independent shaper implementations (IS and FT) can behave in the same incorrect way, then this must point to HAL code as the problem. Is there a testbench, h/w or s/w based, that can allow for these experiments to be carried out in a more efficient way? I'm used to working in a world of simulators and h/w emulators to debug problems like this, so please tell me what the developers of this can do to isolate the problem?

patovers, 2026-03-27:
You have the configs... Attached to previous comments. The problem is that with this particular board, there is no working commit.

thinkyhead, 2026-03-29:
Merging the PR automatically closed this issue. Please comment if you discover further issues even after the timer patch.

thinkyhead, 2026-10-08:
Thanks for all the testing, especially @patovers' finding that the slipping follows the FYSETC drivers rather than the axis. That points at how those modules handle the very quick direction-change-then-step sequences that input shaping produces, which plain motion never does. Could you put a FYSETC driver back on Y and, on current bugfix-2.1.x, try (1) `M114 D` (needs `M114_DETAIL`) after a shifted print to see whether the step counts still match, and (2) adding `#define MINIMUM_STEPPER_PRE_DIR_DELAY 1000`, `#define MINIMUM_STEPPER_POST_DIR_DELAY 1000` and `#define MINIMUM_STEPPER_PULSE_NS 1000` to Configuration_adv.h? Whichever one fixes it tells us exactly what to change for 2.1.3.