MarlinFirmware/Marlin #27989: [BUG] 2.1.2.5 With Bugfixes, Bugfix 2.1.x, Stock cr10 Display wont compile

eclivic1, 2025-07-23:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

No matter what I try, Everytime I use Stock cr10 display it will not compile.

`// Choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_BTT_SKR_V1_3
#endif
`
`[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = LPC1768
include_dir  = Marlin`

`// Connect to EXP1 on RAMPS and compatible boards.
//
#define `CR10_STOCKDISPLAY`
 
`Processing LPC1768 (board: nxp_lpc1768; platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x (0.1.3) > NXP LPC1768
HARDWARE: LPC1768 100MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES:
 - framework-arduino-lpc176x @ 0.2.9
 - toolchain-gccarmnoneeabi @ 1.100301.220327 (10.3.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 3 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Servo @ 1.0.0
|-- U8glib-HAL @ 0.5.4
Building in release mode
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\HAL.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\HAL_SPI.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\MarlinSerial.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\MinSerial.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\eeprom\eeprom_flash.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\eeprom\eeprom_sdcard.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\eeprom\eeprom_wired.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\eeprom_flash.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\eeprom_sdcard.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\eeprom_wired.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\fast_pwm.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\include\digipot_mcp4451_I2C_routines.c.o
cc1.exe: warning: command-line option '-Wno-volatile' is valid for C++/ObjC++ but not for C
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\include\i2c_util.c.o
cc1.exe: warning: command-line option '-Wno-volatile' is valid for C++/ObjC++ but not for C
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\timers.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\u8g\LCD_I2C_routines.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_hw_spi.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_st7920_hw_spi.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_st7920_sw_spi.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_sw_spi.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\usb_serial.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\shared\Delay.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\shared\HAL.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\shared\MinSerial.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\shared\backtrace\backtrace.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\shared\backtrace\unwmemaccess.cpp.o
[... 324 lines cut ...]
SKR 1.3

### LCD/Controller

Stock CR10

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/user-attachments/files/21396396/Marlin.zip)

ellensp, 2025-07-24:
Provided files build without issues.

ellensp, 2025-07-24:
You have a mix of old and new code

These are part of bugfix 2.1.x (note the u8g subdirectory)

\src\lcd\dogm\u8g\u8g_dev_ssd1306_sh1106_128x64_I2C.cpp.o
\src\lcd\dogm\u8g\u8g_dev_ssd1306_sh1106_128x64_SWSPI.cpp.o
\src\lcd\dogm\u8g\u8g_dev_ssd1309_12864.cpp.o
\src\lcd\dogm\u8g\u8g_dev_st7565_64128n_HAL.cpp.o
\src\lcd\dogm\u8g\u8g_dev_st7920_128x64_HAL.cpp.o
\src\lcd\dogm\u8g\u8g_dev_tft_upscale_from_128x64.cpp.o
\src\lcd\dogm\u8g\u8g_dev_uc1701_mini12864_HAL.cpp.
\src\lcd\dogm\u8g\ultralcd_st7920_u8glib_rrd_AVR.cpp.o

These are part of 2.1.2.5
\src\lcd\dogm\u8g_dev_ssd1306_sh1106_128x64_I2C.cpp.o
\src\lcd\dogm\u8g_dev_ssd1309_12864.cpp.o
\src\lcd\dogm\u8g_dev_st7565_64128n_HAL.cpp.o
\src\lcd\dogm\u8g_dev_st7920_128x64_HAL.cpp.o
\src\lcd\dogm\u8g_dev_tft_upscale_from_128x64.cpp.o
\src\lcd\dogm\u8g_dev_uc1701_mini12864_HAL.cpp.o
\src\lcd\dogm\u8g_fontutf8.cpp.o
\src\lcd\dogm\ultralcd_st7920_u8glib_rrd_AVR.cpp.o

You should NOT have both!

You should not extract new Marlin over an old Marlin as it leaves old files. 
Extract Marlin into a new empty directory.