MarlinFirmware/Marlin #27999: [BUG] (bug summary)Linear advance with k factor >0 stops print within seconds

pieterv1010, 2025-08-04:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Linear advance is enabled in configuration_adv, where it is set to 0.

When I set a non-xero value in start G-code the print starts, but halts within seconds (no error shown).

### Bug Timeline

today

### Expected behavior

I expected the print to proceed

### Actual behavior

As above: print halts after few seconds.

### Steps to Reproduce

1.set non-zero k value in slicer
2. slice
3. start print

### Version of Marlin Firmware

2.1.3.b3

### Printer model

custom coreXY

### Electronics

re-arm board

### LCD/Controller

discount lcd

### Bed Leveling

None

### Your Slicer

None

### Host Software

OctoPrint

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/user-attachments/files/21583634/Configuration.zip)

thisiskeithb, 2025-08-06:
> Gcode files not allowed

Compress the gcode into a zip and you’ll be able to attach it.

pieterv1010, 2025-08-06:
[casebody.zip](https://github.com/user-attachments/files/21619328/casebody.zip)

Thanks, here it is.

classicrocker883, 2025-08-09:
> It's a re-arm board, so quite old.

this is probably the issue. incompatible boards will freeze. I recall trying to get a Creality V422 board with linear advance, and wasnt fixed until someone was able to make it work via source code/firmware.

thinkyhead, 2026-10-12:
We ran your casebody G-code in the simulator with classic Linear Advance and your steps/mm, at K2 and at lower K values, and it printed through the skirt and into the part without halting. This isn't related to the Smooth LA problems in #28264, since your config doesn't enable Smooth LA.

Note that K2 is far above the normal range for current Linear Advance (typically 0–0.2). Please retest current `bugfix-2.1.x` with a realistic value like `M900 K0.05`. If it still stops, tell us whether the board resets or freezes, and post the serial output with `LA_DEBUG` enabled and `M111 S2` set.