MarlinFirmware/Marlin #28002: [BUG] Simulator - `warning: writing 1 byte into a region of size 0`

classicrocker883, 2025-08-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Compiling for Marlin Simulator gives the following warning message:

```yml
Marlin/src/module/temperature.cpp: In static member function ‘static void Temperature::PID_autotune(celsius_t, heater_id_t, int8_t, bool)’:
Marlin/src/module/temperature.cpp:838:155: warning: writing 1 byte into a region of size 0 [-Wstringop-overflow=]
  838 |     #define SHV(V) C_TERN(ischamber, temp_chamber.soft_pwm_amount = V, B_TERN(isbed, temp_bed.soft_pwm_amount = V, temp_hotend[heater_id].soft_pwm_amount = V))
      |                                                                                                                                                           ^
Marlin/src/module/temperature.cpp:838:72: note: in expansion of macro ‘B_TERN’
  838 |     #define SHV(V) C_TERN(ischamber, temp_chamber.soft_pwm_amount = V, B_TERN(isbed, temp_bed.soft_pwm_amount = V, temp_hotend[heater_id].soft_pwm_amount = V))
      |                                                                        ^~~~~~
Marlin/src/module/temperature.cpp:953:11: note: in expansion of macro ‘SHV’
  950 |           SHV((bias + d) >> 1);
      |           ^~~
Marlin/src/module/temperature.cpp:915:11: note: in expansion of macro ‘SHV’
  915 |           SHV((bias - d) >> 1);
      |           ^~~
Marlin/src/module/temperature.cpp:340:17: note: at offset [-78, -34] into destination object ‘Temperature::temp_hotend’ of size 44
  340 |   hotend_info_t Temperature::temp_hotend[HOTENDS];
```

### Bug Timeline

Probably as old as time, but noticed recently

### Expected behavior

No warnings, as it does with other Configurations

### Actual behavior

Only when compiling for Marlin Simulator gives this warning.

### Steps to Reproduce

1. Set `env` to **simulator_linux_release** (haven't tried Windows or MacOS)
2. Upload the Simulator configs: [**Configuration.h**](https://github.com/MarlinFirmware/Configurations/blob/bugfix-2.1.x/config/examples/Simulator/Configuration.h) -- [**Configuration_adv.h**](https://github.com/MarlinFirmware/Configurations/blob/bugfix-2.1.x/config/examples/Simulator/Configuration_adv.h)

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Sim

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

---

## Possible fix?
```diff
-                                                               
    #define GHV(C,B,H) C_TERN(ischamber, C, B_TERN(isbed, B, H))
-   #define SHV(V) C_TERN(ischamber, temp_chamber.soft_pwm_amount = V, B_TERN(isbed, temp_bed.soft_pwm_amount = V, temp_hotend[heater_id].soft_pwm_amount = V))
_______________________________________________

+   #define H_TERN(A) ((heater_id >= 0 && heater_id < HOTENDS) ? (A) : 0)
    #define GHV(C,B,H) C_TERN(ischamber, C, B_TERN(isbed, B, H))
+   #define SHV(V) C_TERN(ischamber, temp_chamber.soft_pwm_amount = V, B_TERN(isbed, temp_bed.soft_pwm_amount = V, H_TERN(temp_hotend[heater_id].soft_pwm_amount = V)))
```

ellensp, 2025-08-07:
Error is not present with stock Simulator Configs https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Simulator

ellensp, 2025-08-07:
I'm using git so 

git branch
```
  bugfix-2.0.x
* bugfix-2.1.x   <- shows im using bugfix 2.1.x)
  lts-2.1.2
  release-2.1.3-beta3
```
git pull 
```
   "Already up to date." <- so my local coode is latest code
```
./buildroot/bin/use_example_configs Simulator   
```  restore_configs not found, skipping.   (not needed alread have default configs)
  Fetching examples/Simulator configurations from bugfix-2.1.x...  
```
rm -rf .pio   <--- purge anything cached

<details><summary>Output, does not contain your warnings:</summary>
<p>

pio run -e simulator_linux_release
Processing simulator_linux_release (platform: native; framework: )
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing https://github.com/p3p/MarlinSimUI/archive/af62611296.zip
Unpacking  [####################################]  100%
Library Manager: MarlinSimUI@0.1.0 has been installed!
Library Manager: Installing https://github.com/p3p/Adafruit_NeoPixel/archive/c6b319f447.zip
Unpacking  [####################################]  100%
Library Manager: Adafruit NeoPixel@1.5.0 has been installed!
Library Manager: Installing https://github.com/p3p/LiquidCrystal/archive/322fb5fc23.zip
Unpacking  [####################################]  100%
Library Manager: LiquidCrystal@1.0.0 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ off
Library Manager: Installing marlinfirmware/U8glib-HAL @ 0.5.4
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: U8glib-HAL@0.5.4 has been installed!
Found 4 compatible libraries
Scanning dependencies...
Dependency Graph
|-- MarlinSimulator @ 0.1.0
|-- Adafruit NeoPixel @ 1.5.0
|-- LiquidCrystal @ 1.0.0
|-- U8glib-HAL @ 0.5.4
Building in release mode
Compiling .pio/build/simulator_linux_release/src/src/HAL/NATIVE_SIM/HAL.o
Compiling .pio/build/simulator_linux_release/src/src/HAL/NATIVE_SIM/Servo.o
Compiling .pio/build/simulator_linux_release/src/src/HAL/NATIVE_SIM/pinsDebug.o
Compiling .pio/build/simulator_linux_release/src/src/HAL/NATIVE_SIM/u8g/LCD_I2C_routines.o
Compiling .pio/build/simulator_linux_release/src/src/HAL/NATIVE_SIM/u8g/LCD_pin_routines.o
Compiling .pio/build/simulator_linux_release/src/src/HAL/NATIVE_SIM/u8g/u8g_com_st7920_sw_spi.o
Compiling .pio/build/simulator_linux_release/src/src/HAL/NATIVE_SIM/u8g/u8g_com_sw_spi.o
Compiling .pio/build/simulator_linux_release/src/src/HAL/shared/Delay.o
Compiling .pio/build/simulator_linux_release/src/src/HAL/shared/HAL.o
Compiling .pio/build/simulator_linux_release/src/src/HAL/shared/MinSerial.o
Compiling .pio/build/simulator_linux_release/src/src/HAL/shared/eeprom_api.o
Compiling .pio/build/simulator_linux_release/src/src/HAL/shared/esp_wifi.o
Compiling .pio/build/simulator_linux_release/src/src/HAL/shared/servo.o
Compiling .pio/build/simulator_linux_release/src/src/MarlinCore.o
Compiling .pio/build/simulator_linux_release/src/src/core/serial.o
Compiling .pio/build/simulator_linux_release/src/src/core/utility.o
Compiling .pio/build/simulator_linux_release/src/src/feature/babystep.o
Compiling .pio/build/simulator_linux_release/src/src/feature/bedlevel/abl/bbl.o
Compiling .pio/build/simulator_linux_release/src/src/feature/bedlevel/bedlevel.o
Compiling .pio/build/simulator_linux_release/src/src/feature/bltouch.o
Compiling .pio/build/simulator_linux_release/src/src/feature/cancel_object.o
Compiling .pio/build/simulator_linux_release/src/src/feature/fwretract.o
[... 337 lines cut ...]
Compiling .pio/build/simulator_linux_release/lib3d7/U8glib-HAL/clib/u8g_pbxh24.o
Compiling .pio/build/simulator_linux_release/lib3d7/U8glib-HAL/clib/u8g_polygon.o
Compiling .pio/build/simulator_linux_release/lib3d7/U8glib-HAL/clib/u8g_rect.o
Compiling .pio/build/simulator_linux_release/lib3d7/U8glib-HAL/clib/u8g_rot.o
Compiling .pio/build/simulator_linux_release/lib3d7/U8glib-HAL/clib/u8g_scale.o
Compiling .pio/build/simulator_linux_release/lib3d7/U8glib-HAL/clib/u8g_state.o
Compiling .pio/build/simulator_linux_release/lib3d7/U8glib-HAL/clib/u8g_u16toa.o
Compiling .pio/build/simulator_linux_release/lib3d7/U8glib-HAL/clib/u8g_u8toa.o
Compiling .pio/build/simulator_linux_release/lib3d7/U8glib-HAL/clib/u8g_virtual_screen.o
Archiving .pio/build/simulator_linux_release/lib3d7/libU8glib-HAL.a
Indexing .pio/build/simulator_linux_release/lib3d7/libU8glib-HAL.a
Archiving .pio/build/simulator_linux_release/libd83/libMarlinSimUI.a
Indexing .pio/build/simulator_linux_release/libd83/libMarlinSimUI.a
Linking .pio/build/simulator_linux_release/MarlinSimulator
lto-wrapper: warning: using serial compilation of 30 LTRANS jobs
lto-wrapper: note: see the ‘-flto’ option documentation for more information
======================================================================= [SUCCESS] Took 61.04 seconds =======================================================================

Environment              Status    Duration
-----------------------  --------  ------------
simulator_linux_release  SUCCESS   00:01:01.042
======================================================================== 1 succeeded in 00:01:01.042 ========================================================================

</p>
</details>

classicrocker883, 2025-08-08:
> Error is not present with stock Simulator Configs https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Simulator

one thing I forgot to mention - it doesn't want to work on my Chromebook using Crostini Debian 13 linux.  
so I had to go in **native.ini** and change this line:
```diff
-release_flags     = -g0 -O3 -flto

+release_flags     = -g0 -O3 -fno-lto
```

trying it on a windows using WSL, i can confirm this compiles correctly without that change, and the warning doesnt appear

classicrocker883, 2025-08-08:
one more warning which appears using WSL:

```yml
.pio/libdeps/simulator_linux_release/MarlinSimUI/src/MarlinSimulator/marlin_hal_impl/eeprom.cpp: In static member function ‘static bool PersistentStore::access_start()’:
.pio/libdeps/simulator_linux_release/MarlinSimUI/src/MarlinSimulator/marlin_hal_impl/eeprom.cpp:58:10: warning: ignoring return value of ‘size_t fread(void*, size_t, size_t, FILE*)’ declared with attribute ‘warn_unused_result’ [-Wunused-result]
   58 |     fread(buffer, sizeof(uint8_t), sizeof(buffer), eeprom_file);
```

---

and now trying using just Windows, and `simulator_windows`, it will not compile and gives me a straight error:
```yml
Marlin\src\HAL\NATIVE_SIM\HAL.cpp:43:12: fatal error: HAL.h: No such file or directory

*************************************************************
* Looking for HAL.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:HAL.h"
* Web  > https://registry.platformio.org/search?q=header:HAL.h
*
*************************************************************

   43 |   #include "HAL.h"
      |            ^~~~~~~
compilation terminated.
Compiling .pio\build\simulator_windows\debug\src\src\HAL\NATIVE_SIM\Servo.o
*** [.pio\build\simulator_windows\debug\src\src\HAL\NATIVE_SIM\HAL.o] Error 1
Marlin\src\HAL\NATIVE_SIM\Servo.cpp:22:10: fatal error: ../platforms.h: No such file or directory
   22 | #include "../platforms.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\simulator_windows\debug\src\src\HAL\NATIVE_SIM\Servo.o] Error 1
```

classicrocker883, 2025-08-08:
cannot be a Marlin issue if it *does* work, just not on this machine. 

dont know why but it works fine on another windows machine with simulator_windows, just not that one. I tried a few things, even a work around by including the header files directory within native.ini build_flags, it continued on until missing other header errors. so I tested making `#include "../gcode.h"` to `<gcode.h> for example, got me a bit further trying to see what else it could be, but still a brick wall.

removed .platformio, which was suggested. so, next steps reinstall vscode/platformio. i swear it worked and compiled correctly before. not sure what I'm missing. would hate to have to reinstall like python/msys2...maybe its a dot.net thing?  
I did update everything i can think of.  

out of ideas for that.

---

but the warning for Linux builds using `-fno-lto` - shouldn't be much of an issue?  
if i could compile using the original value but that is its own thing, wouldn't know where to begin how to make that work.

ellensp, 2025-08-08:
delete .pio under marlin.  This contains already downloaded libraries and previous builds. platformio will just recreate it fresh.