MarlinFirmware/Marlin #28010: [BUG] Ender 3 V2 GD32F303RET6 No Display

g3gant, 2025-08-13:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Ender 3 V2 stays on boot screen after flashing Marlon FW.
I've tried to build and flash latest 2.1.3 b3 and the 2.1.x bugfix. Result the same.

Ender 3 V2, the board 4.2.2 on GD32F303RET6, the display is TJC (with original Creality FW), also have BLTouch.

I took one of the tested configurations for Ender 3 v2 v4.2.2, updated the board to `BOARD_CREALITY_V422_GD32_MFL` as well as Serial Port set to 0. For LCD config I picked `DWIN_CREALITY_LCD`.

Not sure what stepper drivers I have (no any marks on the board), so I left TMC2208.

On attempt to flash via SD card, it just stays on the boot screen.

### Expected behavior

The printer works, display shows information

### Actual behavior

Display stays on boot screen

### Steps to Reproduce

Build the attached configuration files with 2.1.x bugfix (or 2.1.3 b3) branch

### Version of Marlin Firmware

---

### Printer model

Ender-3 V2

### Electronics

stock (GD32F303RET6)

### LCD/Controller

TJC

### Other add-ons

BLTouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

SD Card (headless)

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/user-attachments/files/21760237/Configuration.zip)

[image]

ellensp, 2025-08-14:
With build env GD32F303RE_creality_mfl  Serial ports start from 0 vs stm32 where they start from 1

This includes SERIAL_PORT and the screen serial port

When you built marlin you got this warning #warning "Note: Auto-assigned LCD_SERIAL_PORT."

The default port is 3 for the stm32 based board
You need 2

Ie you need to add #define LCD_SERIAL_PORT 2  to your Configuration.h

g3gant, 2025-08-14:
That makes sense. Thanks for taking care of it!

ellensp, 2025-09-07:
note: https://github.com/MarlinFirmware/Marlin/issues/27775  is still outstanding  this stops the lcd from working.

xXHenneBXx, 2026-09-03:
Seeing how this PR is still open id like to make some comments, because I have a TJC ive been going down the rabbit with wanting to build a TJC firmware which led me to want to update marlin, which led me to proui not working yata yat yata yata ..  and im going down the rabbit hole with this PR and well TJC screens itself !  so as stated  by @rh101 it can bypass with STM32F1's Maple variant and run, but maples environment uses deprecated HALs and other libs that are not inside of our current HALs for GD32, some and many needed are still missing, even more so when compiling NO EEPROM is being compiled, even when force enabling them in the configuration files. This below is the compile for the BOARD_CREALITY_V422_GD32_MFL
|-- U8glib-HAL @ 0.5.5
|-- FreeWatchdogTimer @ 1.0.0
|-- SoftwareSerial @ 1.0.0
|-- Wire @ 1.0.0
|-- Servo @ 1.0.0

Tried following each guide above as stated by other Devs  such changing lcd and serial ports did not work.  I DID  however get a working a MFL variant without the need of Maple.. so i had made my own board env in the gd32.ini and added it to pins.h .. 

** NOTE** I copied over Marlin-2.1.2.8(stable) STM32F1 HAL's over so im unsure if this had any affect or not, I was doing testing with HAL's previously and forgot to replace them back in, but any how it works still !

in the ini you make an env as such 

[env:GD32F303RE_creality]
extends                     = env:STM32F103RE_creality
board_build.offset          = 0x7000
board_upload.offset_address = 0x08007000
build_flags                 = ${STM32F103Rx_creality.build_flags}
							  -DMCU_GD32F103RE
                              #-DHXTAL_VALUE=8000000
                              #-DHXTAL_STARTUP_TIMEOUT=0x0500
                              -Os -Wall -ffunction-sections -fdata-sections
                              -Wno-deprecated-declarations
board_build.rename          = update_{date}_{time}.bin

then ensure you set your env in the platformIO.ini .. wait for platformIO to refresh, and click the build, you now have a native STM32F1 compile vs Maple deprecated .. 

and again, this is coming down to HAL's and sort of serial.  It's mentioned on the git for Arduino STM, that there is a breaking change with USART/UART, Notably GD32F1 and GD32F3 chip series use different ports. Doesn't matter if we fixed the pinouts or make a whole board for it.. it wont change it and it still wont work, why because there is no EEPROM being built therefore the flash process is being stopped before actually flashing the firmware to board.  you can see it start to initialize then stops, it does not proceed with the flash, because theres nothing to flash, so basically what it does is just flash a bare blank eeprom not the whole firmware, which then loads NOTHING !!

not even a single serial com command or output, you wont get anything on the screen because the serials arent talking at all.  Because the eeprom is blank.  when building STM32 variants .. you can clearly see an EEPROM being built !! .. 

I firmly believe more or less the issue at hand with this even though @bmourit made a new package for GD32, its a little broken still, if you could fix it that'd be great and testing could be done further to prove this theory, or somehow force an eeprom to be built, but i have not succeeded in doing so !  

Disabling any screen type and using the emulated eeprom, i was able to get a  half successful boot, serial was showing every roughly 2 seconds cosntantly " wait " and never any boot, which confirms this inside of configuration_adv.h noted below

/**
 * Bad Serial-connections can miss a received command by sending an 'ok'
 * Therefore some clients abort after 30 seconds in a timeout.
 * Some other clients start sending commands while receiving a 'wait'.
 * This "wait" is only sent when the buffer is empty. 1 second is a good value here.
 */
//#define NO_TIMEOUTS  1000

// Some clients will have this feature soon. This could make the NO_TIMEOUTS unnecessary.
#define ADVANCED_OK 

NOW most of us probably are building off of ProUI, which ProUI also calls in the use of ui_api from extui.  Inside of ui_api.h this is in the very beginning of its namespace 

namespace ExtUI {

  // The ExtUI implementation can store up to this many bytes
  // in the EEPROM when the methods onStoreSettings and
  // onLoadSettings are called.

  #ifndef EXTUI_EEPROM_DATA_SIZE
    #define EXTUI_EEPROM_DATA_SIZE 48
  #endif
  static constexpr size_t eeprom_data_size = EXTUI_EEPROM_DATA_SIZE;

Could this be overwritting the inital HAL's eeproms ? and gcc isnt strict enough to catch it and slips it through the compile without error ? 
also to support my validation of working native compile with GD32MFL with the changed env to native STM32F1  photos below 
P.S I like the new graph visuals !!

this is my output at USB serial

--- Terminal on COM15 | 115200 8-N-1 --- Available filters and text transformations: debug, default, direct, hexlify, log2file, nocontrol, printable, send_on_enter, time --- More details at https://bit.ly/pio-monitor-filters --- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H 23:41:05.633 > This Boot use for GD32F30XX... 23:41:05.639 > System Boot Version:V1.0.0.2
23:41:05.639 > System Boot Make Time: 11:25:19:Nov 14 2022
23:41:05.642 > Contant us:www.creality.com

[image]
[image]
[image]
[image]

[... cut ...] ade to capture the raw output the serial from the screen/mcu and captures its live events straight from boot to handshake to loading each page and click and movement, each postiion, wether its an icon a font etc .. able to convert some or most of opcodes, as well as capturing the hex output .. 

havent tested this yet but i jsut thought of it not too long ago was to test and see what hex to text would come out with, with the output of info captured.  I use my Raspberry Pi pico2 to capture this !

closed 2026-10-30 by thinkyhead