MarlinFirmware/Marlin #28015: [BUG] `FLY_D5] Compiling issues and not getting any binary file to upload

Amachete, 2025-08-16:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have just swap my old RAMPS, with this cheap board, I thought it would easy......

I have Bltouch, rerap_LCD_PANEL, etc....In the end i got to  compile it.....but i could not create the binary

[image]

I made some changes cherry picking from RAMPS.h to pins_FLY_D5.h

If someone can help  me to finish the compilation would be nice. This is my first experience compiling and uploading a STM32 board.

If you need more information, let me know....I am rewiring the machine, so in the mean time,....i will try to fix this problem.

Thank you in advance. 

### Expected behavior

Compiling and uplad to the board

### Actual behavior

It does not compile, or create binary.

### Version of Marlin Firmware

latest

### Printer model

custom

### Electronics

fly_d5

### LCD/Controller

reprap_SAMRT_CONTROLLER

### Other add-ons

bltouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Changes on PINS_FLY_D5

[Marlin.zip](https://github.com/user-attachments/files/21811528/Marlin.zip)

ellensp, 2025-08-16:
The chip on this controller is tiny.  
STM32F072RB  128 KB of Flash memory and 16 KB, this is half the flash of a 15 year old Mega2560 such as a ramps.

Amachete, 2025-08-16:
OK.....Thanks for the quick response...I did not read carefully the datasheet....