MarlinFirmware/Marlin #28032: [BUG] (bug summary)Marlin-2.1.2.5 version enables MINIPANEL error

mhzx, 2025-09-02:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hello, I am using Marlin-2.1.2.5 version, and I am using an Arduino AVR 2560 motherboard. When I enabled MINIPANEL, I reported the error Compilation error: 'DIOAUX2_08_PIN_RPORT' was not declared in this scope. How should I solve it?

### Version of Marlin Firmware

2.1.2.5

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration_adv.zip](https://github.com/user-attachments/files/22088057/Configuration_adv.zip)

ellensp, 2025-09-02:
Please don't modify the Configuration.h like this. It breaks building the firmware on platformio.   
The code expect to be able to look up the motherboard string in pins.h 

You should also use platformio under vscode, it is 200% faster than Arduino ide.  

In Marlin/src/pins/ramps/pins_RAMPS.h is 
```
    #elif ENABLED(MINIPANEL)

      #ifndef BEEPER_PIN
        #define BEEPER_PIN           AUX2_08_PIN

```

it should say         #define BEEPER_PIN           AUX2_08

This is a typo.

mhzx, 2025-09-02:
[image]

[image]

> 哦，这不是真正的迷你面板液晶显示器，而是一块带兼容屏幕的主板。无论如何，谢谢。

thisiskeithb, 2025-09-03:
That looks *really close* to the Wanhao i3 Mini 0ne+ motherboard (`BOARD_WANHAO_ONEPLUS`) / `MINIPANEL` combo:

|  Front | Back |
|---|---|
| [image] | [image] |

mhzx, 2025-09-03:
> 请不要像这样修改 Configuration.h 文件。这会破坏在 platformio 上构建固件的过程。 代码应该能够在 pins.h 文件中查找主板字符串。
> 
> 您还应该在 vscode 下使用 platformio，它比 Arduino ide 快 200%。
> 
> 该错误是由 Marlin/src/pins/ramps/pins_RAMPS.h 中的拼写错误引起的
> 
>     #elif ENABLED(MINIPANEL)
> 
>       #ifndef BEEPER_PIN
>         #define BEEPER_PIN           AUX2_08_PIN
> 应该说 `#define BEEPER_PIN AUX2_08`

How to set the default contrast ratio?

ellensp, 2025-09-03:
default is set according to lcd type in code

```cpp
#elif ENABLED(MINIPANEL)
  #define _LCD_CONTRAST_INIT 150
```

you adjust it vsa lcd or gcode https://marlinfw.org/docs/gcode/M250.html

ellensp, 2025-09-03:
this value is stored in eeprom when eeprom was initialized or updated,  updating it in code will not work (unless you have EEPROM_INIT_NOW)

mhzx, 2025-09-03:
> right so what you are saying is the lcd is not really a MINIPANEL, but its close so they hacked u8g_dev_uc1701_mini12864.c to make it work.
> 
> comparing original https://github.com/olikraus/u8glib/blob/master/csrc/u8g_dev_uc1701_mini12864.c with https://github.com/inov3d/Aladdinbox-Skycube/blob/master/Stock%20main%20board/u8g_dev_uc1701_mini12864.c
> 
> you get these changes
> 
> diff --git a/csrc/u8g_dev_uc1701_mini12864.c b/csrc/u8g_dev_uc1701_mini12864.c
> index 489c60c..f856824 100644
> --- a/csrc/u8g_dev_uc1701_mini12864.c
> +++ b/csrc/u8g_dev_uc1701_mini12864.c
> @@ -51,13 +51,13 @@ static const uint8_t u8g_dev_uc1701_mini12864_init_seq[] PROGMEM = {
>    0x0a0,		/* ADC set to reverse */
>    0x0c8,		/* common output mode */
>    0x0a6,		/* display normal, bit val 0: LCD pixel off. */
> -  0x0a2,		/* LCD bias 1/9 */
> +  0x0a3,		/* LCD bias 1/9 */
>    0x02f,		/* all power  control circuits on */
>    0x0f8,		/* set booster ratio to */
>    0x000,		/* 4x */
>    0x023,		/* set V0 voltage resistor ratio to large */
>    0x081,		/* set contrast */
> -  0x027,		/* contrast value */
> +  0x032,		/* contrast value */
>    0x0ac,		/* indicator */
>    0x000,		/* disable */
>    0x0af,		/* display on */
> @@ -79,24 +79,6 @@ static const uint8_t u8g_dev_uc1701_mini12864_data_start[] PROGMEM = {
>    U8G_ESC_END                /* end of sequence */
>  };
>  
> -static const uint8_t u8g_dev_uc1701_mini12864_sleep_on[] PROGMEM = {
> -  U8G_ESC_ADR(0),           /* instruction mode */
> -  U8G_ESC_CS(1),            /* enable chip */
> -  0x0ae,                    /* display off */
> -  0x0a5,                    /* enter powersafe: all pixel on */
> -  U8G_ESC_CS(0),            /* disable chip */
> -  U8G_ESC_END               /* end of sequence */
> -};
> -
> -static const uint8_t u8g_dev_uc1701_mini12864_sleep_off[] PROGMEM = {
> -  U8G_ESC_ADR(0),           /* instruction mode */
> -  U8G_ESC_CS(1),            /* enable chip */
> -  0x0a4,                    /* display on */
> -  0x0af,                    /* all pixel off */
> -  U8G_ESC_CS(0),            /* disable chip */
> -  U8G_ESC_END               /* end of sequence */
> -};
> -
>  uint8_t u8g_dev_uc1701_mini12864_fn(u8g_t *u8g, u8g_dev_t *dev, uint8_t msg, void *arg)
>  {
>    switch(msg)
> @@ -125,12 +107,6 @@ uint8_t u8g_dev_uc1701_mini12864_fn(u8g_t *u8g, u8g_dev_t *dev, uint8_t msg, voi
>        u8g_WriteByte(u8g, dev, (*(uint8_t *)arg) >> 2);
>        u8g_SetChipSelect(u8g, dev, 0);      
>        return 1;
> -    case U8G_DEV_MSG_SLEEP_ON:
> -      u8g_WriteEscSeqP(u8g, dev, u8g_dev_uc1701_mini12864_sleep_on);
> -      return 1;
> -    case U8G_DEV_MSG_SLEEP_OFF:
> -      u8g_WriteEscSeqP(u8g, dev, u8g_dev_uc1701_mini12864_sleep_off);
> -      return 1;
>    }
>    return u8g_dev_pb8v1_base_fn(u8g, dev, msg, arg);
>  }
> @@ -169,12 +145,6 @@ uint8_t u8g_dev_uc1701_mini12864_2x_fn(u8g_t *u8g, u8g_dev_t *dev, uint8_t msg,
>        u8g_WriteByte(u8g, dev, (*(uint8_t *)arg) >> 2);
>        u8g_SetChipSelect(u8g, dev, 0);      
>        return 1;
> -    case U8G_DEV_MSG_SLEEP_ON:
> -      u8g_WriteEscSeqP(u8g, dev, u8g_dev_uc1701_mini12864_sleep_on);
> -      return 1;
> -    case U8G_DEV_MSG_SLEEP_OFF:
> -      u8g_WriteEscSeqP(u8g, dev, u8g_dev_uc1701_mini12864_sleep_off);
> -      return 1;
>    }
>    return u8g_dev_pb16v1_base_fn(u8g, dev, msg, arg);
>  }
> In modern marlin the changes they made are equivalent of editing
> 
> Marlin/src/lcd/dogm/u8g/u8g_dev_uc1701_mini12864_HAL.cpp
> 
> UC1701_BIAS_MODE(0), // LCD bias 1/9 -> UC1701_BIAS_MODE(1), // LCD bias 1/9 UC1701_CONTRAST(0x27), // Set contrast -> UC1701_CONTRAST(0x32), // Set contrast
> 
> ie
> 
> diff --git a/Marlin/src/lcd/dogm/u8g/u8g_dev_uc1701_mini12864_HAL.cpp b/Marlin/src/lcd/dogm/u8g/u8g_dev_uc1701_mini12864_HAL.cpp
> index 1d1291ff9a..63f43c169b 100644
> --- a/Marlin/src/lcd/dogm/u8g/u8g_dev_uc1701_mini12864_HAL.cpp
> +++ b/Marlin/src/lcd/dogm/u8g/u8g_dev_uc1701_mini12864_HAL.cpp
> @@ -93,11 +93,11 @@ static const uint8_t u8g_dev_uc1701_mini12864_HAL_init_seq[] PROGMEM = {
>    UC1701_ADC_REVERSE(0),      // ADC set to reverse
>    UC1701_OUT_MODE(1),         // Common output mode
>    UC1701_INVERTED(0),         // Display normal, bit val 0: LCD pixel off
> -  UC1701_BIAS_MODE(0),        // LCD bias 1/9
> +  UC1701_BIAS_MODE(1),        // LCD bias 1/9
>    UC1701_POWER_CONTROL(0x7),  // All power control circuits on
>    UC1701_BOOST_RATIO(0x0),    // Set booster ratio to 4x
>    UC1701_V5_RATIO(3),         // Set V0 voltage resistor ratio to large
> -  UC1701_CONTRAST(0x27),      // Set contrast
> +  UC1701_CONTRAST(0x32),      // Set contrast
>    UC1701_INDICATOR(0),        // Indicator disable
>    UC1701_ON(1),               // Display on
> They also remove a bunch of sleep code, but that is not part of the new code

Thank you very much for your help. I have successfully debugged the screen correctly. I adjusted the pins_RAMPS.h file to the following content:
#elif ENABLED(MINIPANEL)
      #define LCD_RESET_PIN     25
      #ifndef BEEPER_PIN
        #define BEEPER_PIN               37//AUX2_08
      #endif
      #define LCD_BACKLIGHT_PIN          AUX2_10

      #define DOGLCD_A0                  27//AUX2_07
      #define DOGLCD_CS                  29//AUX2_09

      #define BTN_EN1                    33//AUX2_06
      #define BTN_EN2                    31//AUX2_04
      #define BTN_ENC                    35//AUX2_03

      #ifndef SD_DETECT_PIN
        #define SD_DETECT_PIN            AUX3_02
      #endif
      #define KILL_PIN                   AUX2_05

[image]

Note: #define LCD_RESET_PIN 25 is a part that does not exist in your code. I will share my experience with you, hoping to help you adapt to more machines!