MarlinFirmware/Marlin #28081: [BUG] `T1` in gcode for single extruder treated as `T0`

classicrocker883, 2025-09-28:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

This is an issue which you can look more into here: [classicrocker883/MRiscoCProUI/issues/197](https://github.com/classicrocker883/MRiscoCProUI/issues/197)
Basically the jist is some slicers will make their gcode and use `T1` for example:
```gcode
M104 S240 T0
M104 S0 T1
```
It's kind of like I guess if you want to print with 1 nozzle, it forces another (if multiple) to not be used.  
For single extruder/single heated nozzle machines, the current code will make `T1` like `T0` and stop heating, breaking the printing process.

I dont know if this is meant to be or just an inconvenience and issue.  
So is maybe setting `T1` and having it treat it as `T0` might be like a redundant fallback safety thing?

A simple quick fix for me was this: add `!` to `HAS_TOOLCHANGE`  
in **gcode.cpp**
```y
int8_t GcodeSuite::get_target_extruder_from_command() {
  #if !HAS_TOOLCHANGE
    if (parser.seenval('T')) {
      const int8_t e = parser.value_byte();
      if (e < EXTRUDERS) return e;
      SERIAL_ECHO_START();
      SERIAL_CHAR('M'); SERIAL_ECHO(parser.codenum);
      SERIAL_ECHOLNPGM(" " STR_INVALID_EXTRUDER " ", e);
      return -1;
    }
  #endif
  return active_extruder;
}
```

### Bug Timeline

Old

### Expected behavior

In a single heated extruder, you'd expect it to run the .gcode file and print normally.

### Actual behavior

The printer reads `T1` or extra extruders as `T0`. This can cause unwanted behavior.

### Steps to Reproduce

1. Have a single extruder machine like Ender-3, Voxelab Aquila
2. Use a slicer like VoxelMaker which produces a .gcode file with `T1`
2b. **OR** Enter `M104 S0 T1` after the _Set Hotend Temperature_/`M104` in start gcode
3. Try printing

### Version of Marlin Firmware

bugfix-2.1.3

### Printer model

Voxelab Aquila

### Bed Leveling

None

### Your Slicer

Other (explain below)

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I suppose you can use any Creality Ender-3 V2 or similar config - one that has single heated nozzle.

ellensp, 2025-09-28:
all g-codes that have a T parameter have this issue
M92, M104, M105, M109, M200, M201, M203, M218, M221, M600, M603, M701 and M702

ellensp, 2025-09-28:
A for how to fix it....

The draconian way would be if any of these gcodes has a T parameter while there is only the 1 tool it should error.
so slicers that produce bad gcode like ffslicer cant do any damage and the user gets a error message

Or we just error when T > extruders..  Ie remove HAS_TOOLCHANGE block from get_target_extruder_from_command

classicrocker883, 2025-09-28:
well it doesn't actually give the user an error message, it continues on; returns the `active_extruder`. I don't know about other Gcode commands.  
based on my proposed change it does say "Invalid Extruder 1" and that sounds like it works like it should.

closed 2025-10-01 by ellensp